greatscottgadgets/cynthion #39: Support amaranth~=0.5

antoinevg, 2024-01-26:
At the time of writing `cynthion` gateware supports `"amaranth~=0.4.0",`.  The next Amaranth release will be 0.5 which will bring some substantial changes: https://github.com/amaranth-lang/amaranth/blob/main/docs/changes.rst#version-05-unreleased

Some of these affect the `cynthion`, `luna` and `luna-soc` gateware directly and, at the time of writing, include:

- [ ] Final deprecation of the `Sample`, `Past`, `Stable`, `Rose` and `Fell` helpers in favour of manually instantiated registers.
  - Commit: https://github.com/amaranth-lang/amaranth/commit/750cbbc3c704cf35aaed9455d5eb80b9deb1adf6
  - Migration Example: https://github.com/amaranth-lang/amaranth/commit/28e1d2833f7c29c367497a48fe84d40c33a98b5f
- [ ] Uses of `m.Case()` with no pattern should be replaced with `m.Default()`
- [ ] `amaranth.hdl.rec.Record` will be deprecated in 0.5 and removed in 0.6
- [ ] `from amaranth.vendor.lattice_ecp5 import LatticeECP5Platform` becomes `from amaranth.vendor import LatticeECP5Platform`

There are also a number of major changes happening in `amaranth-soc` that affect https://github.com/lambdaconcept/lambdasoc, including:

- [ ]  [event: migrate to lib.wiring interfaces.](https://github.com/amaranth-lang/amaranth-soc/commit/feec75180f0ad9fc8c8231933feb8efa26dc9875)
- [ ] [csr.bus: migrate to lib.wiring interfaces. ](https://github.com/amaranth-lang/amaranth-soc/commit/141d1d4b260cfb720bb1467e6db1abc14100bebc)
- [ ] [wishbone.bus: migrate to lib.wiring interfaces.](https://github.com/amaranth-lang/amaranth-soc/commit/c5e1e6606a9452601c274e3e48cda8dbcf353b96)
- [ ] [Remove log2_int](https://amaranth-lang.org/rfcs/0017-remove-log2-int.html#remove-log2_int)

claui, 2025-05-20:
@antoinevg Now that Cynthion supports amaranth~=0.5, is this issue still needed?

antoinevg, 2025-05-21:
Fixed in: https://github.com/greatscottgadgets/cynthion/releases/tag/0.2.0