greatscottgadgets/cynthion #84: docs: Cynthion hardware documentation

antoinevg, 2024-05-13:
We need some basic documentation for the Cynthion hardware.

At minimum we need to have pages that cover the following topics:

* [Introduction](https://hackrf.readthedocs.io/en/latest/hackrf_one.html)
* [Minimum Host System Requirements for Cynthion](https://hackrf.readthedocs.io/en/latest/hackrf_minimum_requirements.html)
* [List of Hardware Revisions](https://hackrf.readthedocs.io/en/latest/list_of_hardware_revisions.html)
* [Hardware Components](https://hackrf.readthedocs.io/en/latest/hardware_components.html)
* Purchasing Information

(Linked to equivalent HackRF docs as an example)

antoinevg, 2024-05-14:
It might also be cool to have a section with explicit rules/recommendations for people who want to build their own hardware. (e.g. something like they have in the Glasgow docs: https://glasgow-embedded.org/latest/build.html)

martinling, 2024-07-09:
Something that would really help is a block diagram similar to the one we have for HackRF.

mossmann, 2024-07-22:
Some information about power switching is needed. For starters, the introduction should be updated to say that CONTROL is the primary port for both control and power.

mossmann, 2024-08-28:
The r1.4.0 hardware design release has been published: https://github.com/greatscottgadgets/cynthion-hardware/releases/tag/r1.4.0