greatscottgadgets/cynthion #105: Is "USB run through" a must? Or can USB D+ and D-  (and GND of course) only can be fed into Cynthion?

capiman, 2024-06-25:
Can you tell me, if USB must run through the device?
E.g. there is input for USB (from USB host) and an output (to USB device).

Or would it be also be possible to just "clamp" on D+ and D-?

Background: I have a board (PCB) which has an USB host and an USB device on it.
If it runs through the device I must find a place where to open the USB bus,
cut it in two halfs and connect one to input and the other side to output.
If Cynthion can be connected be "just two wire" on the PCB it would be easier.
I can assume the "run through" is used for detecting the direction of the USB
signals. Is this the case?

Many thanks for clarification!

martinling, 2024-06-27:
The problem would be signal integrity in High Speed (480Mbps) operation. If you only need Full (12Mbps) or Low (1.5Mbps) operation then there's a good chance it would work okay.

If it's using High Speed though, then it's critical that the signal path is a clean 90 ohm differential transmission line. If you take a branch off an existing pair, you're adding a stub that will create reflections and losses, disrupting the signal. It's very sensitive to this sort of thing.

If you need to tap a sniff a speed signal running across a board, then cutting it somewhere in the middle and attaching cables to each side of the break is the best way to do it, unless you can get your hands on an active differential probe.

closed 2024-07-05 by martinling

capiman, 2024-07-05:
Thank you for the clarification!