greatscottgadgets/cynthion #118: docs: using Cynthion for low-level proxying?

xairy, 2024-07-05:
[Using Cynthion with USB Proxy](https://cynthion.readthedocs.io/en/latest/getting_started_usbproxy.html) shows how to use Cythion for high-level proxying, i.e. when the proxied device is connected to a control host running Facedancer and the device is then reemulated for the target host via Cynthion.

Is it possible to use Cynthion for low-level proxying, i.e. connecting the proxied device to the Target A post and the target host to the Target C port (similar setup as for using Packetry) and then changing the transmitted bits in the fly?

It would be great to see documentation if this is possible or otherwise a note that this is not supported.

Thank you!

martinling, 2024-07-05:
Yes, this is possible - but note that it will involve using the AUX port and _one_ of the TARGET ports, not the two TARGET ports. The data lines of two TARGET ports are directly connected to each other for passive sniffing. To modify traffic requires going through the FPGA, as in the Facedancer case.

Your gateware would need to interface with the AUX and TARGET PHYs as necessary to implement the modifications you want to create. The LUNA gateware library has a lot of the building blocks you'd need to do this, but the details would be up to you.

martinling, 2024-07-18:
Sorry, should have left this open to remind us to add something about this to the docs.