greatscottgadgets/cynthion #130: No Backend error on MAC OS

sb003726, 2024-07-09:
Hi all,
I am trying to test my new cynthion on an iMAC, (Intel, OS X 10.13.6, python 3.12.4), the device is reported in system information:

`USB Analyzer :

  Identifiant du produit :	0x615b
  Identifiant du fournisseur :	0x1d50
  Version :	1.04
  Numéro de série :	273384df30c460de
  Vitesse :	Jusqu’à 480 Mb/s
  Fabricant :	Cynthion Project
  Identifiant de l’emplacement :	0x24100000 / 2
  Courant disponible (mA) :	500
  Courant requis (mA) :	500
  Exploitation supplémentaire actuelle (mA) :	0`

But running "Cynthion info" returns the following error:

`Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.12/bin/cynthion", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/cynthion/commands/cli.py", line 91, in main
    bitstream_info = get_bitstream_information()
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/cynthion/commands/util.py", line 30, in get_bitstream_information
    device = usb.core.find(
             ^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/usb/core.py", line 1309, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available`

Any ideas please on how to debug this? (Is cynthion even supported on OS X?)
Regards
Steve

martinling, 2024-07-09:
I think you're missing libusb, which you can install with Homebrew or Macports, then it should work. We'll update the documentation to reflect this.

On other systems, installing the Python `libusb1` package brings the necessary native library with it, but not on OSX/macOS apparently, so it has to be done separately.

sb003726, 2024-07-09:
That was indeed the problem, installed libusb and all is well.
Thanks for the help

closed 2024-07-09 by martinling

martinling, 2024-07-09:
Thanks for letting us know! I've opened #132 to remind us to update the documentation.