greatscottgadgets/cynthion #131: The `shared` directory is not installed when building on Windows

martinling, 2024-07-09:
When the repository is checked out on Windows with the default Git settings, the symbolic link from `cynthion\python\src\shared` is not reproduced as a symbolic link, so the `shared` directory is not installed.

This results in an error at import time:
```
C:\Users\Martin\Documents\cynthion>cynthion info
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Martin\AppData\Roaming\Python\Python312\Scripts\cynthion.exe\__main__.py", line 4, in <module>
  File "C:\Users\Martin\AppData\Roaming\Python\Python312\site-packages\cynthion\__init__.py", line 16, in <module>
    from .cynthion import Cynthion
  File "C:\Users\Martin\AppData\Roaming\Python\Python312\site-packages\cynthion\cynthion.py", line 5, in <module>
    from .board import CynthionBoard
  File "C:\Users\Martin\AppData\Roaming\Python\Python312\site-packages\cynthion\board.py", line 17, in <module>
    from .shared import usb
ImportError: cannot import name 'usb' from 'cynthion.shared' (C:\Users\Martin\AppData\Roaming\Python\Python312\site-packages\cynthion\shared.py)
```

There is no problem when using the currently released wheels, this only happens when installing from git on Windows.

mossmann, 2024-07-25:
It is possible to enable symlinks on Windows:
https://www.scivision.dev/git-windows-symlink/

antoinevg, 2024-07-25:
Cynthion docs have been updated with instructions: https://cynthion.readthedocs.io/en/latest/developer/introduction.html

martinling, 2024-07-30:
If there isn't a non-symlink solution here, and the resolution is to require developers to enable symlinks on Windows, can we check at build time that this has been done and that the symlink is present?