greatscottgadgets/cynthion #133: usb stick proxy - LIBUSB_ERROR_BUSY

idealdelay, 2024-07-09:
Using a Cynthion and running the facedancer usbproxy.py example. Not sure if issues is FD, Cynthion or myself. It idles until I connect to the target usb host computer, exchanges some information then errors with LIBUSB_ERROR_BUSY. I have tried this with four different usb mass storage devices on two different Ubuntu 20.04 computers.

python3 proxy.py 
INFO    | greatdancer    | Skipping GreatFET-based devices, as the greatfet python module isn't installed.
INFO    | moondancer     | Using the Moondancer backend.
WARNING | moondancer     | Moondancer only supports USB Full and High Speed. Ignoring requested speed: SUPER
INFO    | moondancer     | Connected SUPER speed device 'facedancer.proxy.USBProxyDevice' to target host.
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | logging        | [18:14:23] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=256, index=0, length=64, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x73cf18deba30>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x73cf17f1f190>))
INFO    | logging        | [18:14:23] <: b'\x12\x01\x00\x03\x00\x00\x00@\x81\x07\x81U\x00\x01\x01\x02\x03\x01'
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | logging        | [18:14:23] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=256, index=0, length=18, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x73cf18deba30>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x73cf17f1f190>))
INFO    | logging        | [18:14:23] <: b'\x12\x01\x00\x03\x00\x00\x00@\x81\x07\x81U\x00\x01\x01\x02\x03\x01'
INFO    | logging        | [18:14:23] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=3840, index=0, length=5, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x73cf18deba30>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x73cf17f1f190>))
INFO    | logging        | [18:14:23] <: b'\x05\x0f\x16\x00\x02'
INFO    | logging        | [18:14:23] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=3840, index=0, length=22, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x73cf18deba30>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x73cf17f1f190>))
INFO    | logging        | [18:14:23] <: b'\x05\x0f\x16\x00\x02\x07\x10\x02\x02\x00\x00\x00\n\x10\x03\x00\x0e\x00\x01\n\x00\x01'
INFO    | logging        | [18:14:23] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=512, index=0, length=9, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x73cf18deba30>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x73cf17f1f190>))
INFO    | logging        | [18:14:23] <: b'\t\x02,\x00\x01\x01\x00\x80p'
INFO    | logging        | [18:14:23] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=512, index=0, length=44, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x73cf18deba30>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x73cf17f1f190>))
INFO    | logging        | [18:14:23] <: b'\t\x02,\x00\x01\x01\x00\x80p\t\x04\x00\x00\x02\x08\x06P\x00\x07\x05\x81\x02\x00\x04\x00\x060\x01\x00\x00\x00\x07\x05\x02\x02\x00\x04\x00\x060\x0f\x00\x00\x00'
INFO    | logging        | [18:14:23] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=768, index=0, length=255, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x73cf18deba30>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x73cf17f1f190>))
INFO    | logging        | [18:14:23] <: ̄Љ
INFO    | logging        | [18:14:23] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=770, index=1033, length=255, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_ [... cut ...]
    cls.device_handle.attachKernelDriver(number)
  File "/home/user/.local/lib/python3.10/site-packages/usb1/__init__.py", line 1207, in attachKernelDriver
    mayRaiseUSBError(
  File "/home/user/.local/lib/python3.10/site-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/home/user/.local/lib/python3.10/site-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorBusy: LIBUSB_ERROR_BUSY [-6]

idealdelay, 2024-07-10:
just updated cynthion/bitstream from the changes made yesterday. looks like its working now!

martinling, 2024-07-10:
Great! Thanks for letting us know.

closed 2024-07-10 by martinling