greatscottgadgets/cynthion #134: proxy - usb drive doesn't mount

idealdelay, 2024-07-10:
im running the facedancer simplest usb proxy via a cynthion to proxy a usb storage stick. it seems to be running / logging but the target host computer is not mount the device (I waited 1.5 hours), it is however present in lsusb (on the traget host). the stick mounts as expected when directly connected to the target host. should the device mount as per normal or any idea on how to track down the issue, be it facedancer/cynthion? (im running the newest version of everytning)

python3 proxy.py 
INFO    | greatdancer    | Skipping GreatFET-based devices, as the greatfet python module isn't installed.
INFO    | moondancer     | Using the Moondancer backend.
INFO    | moondancer     | Connected HIGH speed device 'facedancer.proxy.USBProxyDevice' to target host.

INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | standard       | -- Patched device descriptor. --
INFO    | logging        | [11:27:50] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=256, index=0, length=64, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7fca7d3ebb50>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7fca7c536170>))
INFO    | logging        | [11:27:50] <: b'\x12\x01\x00\x02\x00\x00\x00@\x81\x07uU&\x01\x01\x02\x03\x01'
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | standard       | -- Patched device descriptor. --
INFO    | logging        | [11:27:50] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=256, index=0, length=18, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7fca7d3ebb50>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7fca7c536170>))
INFO    | logging        | [11:27:50] <: b'\x12\x01\x00\x02\x00\x00\x00@\x81\x07uU&\x01\x01\x02\x03\x01'
INFO    | logging        | [11:27:50] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=512, index=0, length=9, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7fca7d3ebb50>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7fca7c536170>))
INFO    | logging        | [11:27:50] <: b'\t\x02 \x00\x01\x01\x00\x80d'
INFO    | standard       | -- Storing configuration USBConfiguration(number=1, configuration_string=0, max_power=100, self_powered=0, supports_remote_wakeup=0, parent=None, interfaces={0: USBInterface(name=None, number=0, alternate=0, class_number=8, subclass_number=6, protocol_number=80, interface_string=0, descriptors={<USBDescriptorTypeNumber.INTERFACE: 4>: <bound method USBInterface.get_descriptor of ...>}, class_descriptor=None, endpoints={129: USBEndpoint(number=1, direction=1, transfer_type=2, synchronization_type=0, usage_type=0, max_packet_size=512, interval=0, parent=...), 2: USBEndpoint(number=2, direction=0, transfer_type=2, synchronization_type=0, usage_type=0, max_packet_size=512, interval=1, parent=...)}, parent=None)}) --
INFO    | logging        | [11:27:50] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=512, index=0, length=32, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7fca7d3ebb50>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7fca7c536170>))
INFO    | logging        | [11:27:50] <: b'\t\x02 \x00\x01\x01\x00\x80d\t\x04\x00\x00\x02\x08\x06P\x00\x07\x05\x81\x02\x00\x02\x00\x07\x05\x02\x02\x00\x02\x01'
INFO    | logging        | [11:27:50] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=768, index=0, length=255, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7fca7d3ebb50>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7fca7c536170>))
INFO    | logging        | [11:27:50] <: ̄Љ
INFO    | logging        | [11:27:50] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=770, index=1033, length=255, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7fca7d3ebb50>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7fca7c536170>))
INFO    | logging        | [11:27:50] <: ̚Cruzer Glide
INFO    | logging        | [11:27:50] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=769, index=1033, length=255, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string='Facedancer', product_string='Generic USB Device', serial_number_string='S/N 3420E', supported_languages=(<LanguageIDs.ENGLISH_US: 1033>, [... cut ...] LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=2, device_speed=None, descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7fca7d3ebb50>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7fca7c536170>))

antoinevg, 2024-07-10:
SanDisk Ultra 16GB USB 3.0:

```
[35343.586852] scsi host6: usb-storage 3-3:1.0
[35344.621308] scsi 6:0:0:0: Direct-Access     SanDisk  Ultra USB 3.0    1.00 PQ: 0 ANSI: 6
[35344.621522] sd 6:0:0:0: Attached scsi generic sg2 type 0
[35344.636509] sd 6:0:0:0: [sdb] 30031872 512-byte logical blocks: (15.4 GB/14.3 GiB)
[35344.637164] sd 6:0:0:0: [sdb] Write Protect is off
[35344.637168] sd 6:0:0:0: [sdb] Mode Sense: 43 00 00 00
[35344.637484] sd 6:0:0:0: [sdb] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[35344.641892]  sdb: sdb1
[35344.642036] sd 6:0:0:0: [sdb] Attached SCSI removable disk
```

antoinevg, 2024-07-10:
Also: Sandisk 3.2Gen1 BM2209005782w sdddc3

dominikfehr, 2024-08-06:
can confirm same behavior to Windows 11 target host running (visible in device manager, no mounting)

antoinevg, 2024-08-21:
Can you please try to instead plug the device to be proxied into a High-speed USB 2.0 hub that's plugged into the target host?

antoinevg, 2024-08-28:
Fixed in #183 and https://github.com/greatscottgadgets/facedancer/pull/115

PR #183 (2024-09-16): Fix ZLP bug on Facedancer CONTROL IN transfers
This PR:

* Fixes a subtle bug with the USB HAL driver when sending a ZLP.
* Adds moondancer firmware support for: https://github.com/greatscottgadgets/facedancer/pull/115

Depends on: https://github.com/greatscottgadgets/facedancer/pull/115

---
Closes #71
Closes #134
Closes #156
closed 2024-09-16 by antoinevg