greatscottgadgets/cynthion #135: Cynthion does not connect to hardware

fcbsd, 2024-07-12:
I am running Cynthion on OpenBSD (OpenBSD heather.crowsons.com 7.5 GENERIC.MP#82 amd64) and I am getting the "No Apollo device or stub interface found" when running `cynthion info` , while I appreciate that OpenBSD probably not a supported platform, I would like to know how `cynthion` attaches to the device - looking through the code for `1d50` does not show any obvious place where the connection is made...any hints would be appreciated.

Further info:
```
heather:fred ~: grep "Cynthion" /var/log/messages |tail -1                                                
Jul 12 14:26:31 heather /bsd: ugen2 at uhub2 port 4 "Cynthion Project USB Analyzer" rev 2.00/1.04 addr 9
heather:fred ~: usbdevs -v |grep Cynthion
addr 09: 1d50:615b Cynthion Project, USB Analyzer
heather:fred ~: .local/bin/cynthion info
No Apollo device or stub interface found.
```

martinling, 2024-07-12:
We haven't tested on OpenBSD, and we don't really have the resources to officially support it, but I see no fundamental reason that most of our tools shouldn't work there, as the platform is supported by libusb and the associated Python USB libraries. You will have issues if you want to run [Packetry](https://github.com/greatscottgadgets/packetry) though, because that's using [nusb](https://github.com/kevinmehall/nusb), which doesn't have an OpenBSD backend as yet.

Returning to your specific issue, the code you're looking for is in the [Apollo](https://github.com/greatscottgadgets/apollo) project, the Python side of which is a dependency of the `cynthion` package. You should be able to run the `apollo` command line tool that was installed with it.

Basically, it's looking for either an Apollo device - i.e. the MCU on the Cynthion, which will enumerate if you push the PROGRAM button - or a special "stub" interface that can be exposed by USB devices running from the FPGA, which supports asking the device to hand control back to the MCU.

Why it can't find it will be down to some issue with listing, opening and accessing USB devices and interfaces on them.

fcbsd, 2024-07-12:
thanks I'll take a look