greatscottgadgets/cynthion #136: "No backend available" on macOS

mossmann, 2024-07-13:
`cynthion info` and various other commands error out with:
```
usb.core.NoBackendError: No backend available
```
This happens because pyusb can't find the libusb (dynamically linked C) library. Part of the solution is to ensure that libusb is installed (see #132), but there can also be problems with pyusb finding the installed libusb. This is especially likely if libusb was installed with Homebrew and the user is using Apple's system Python.

One solution is to install Homebrew's Python which knows where to look for Homebrew's libraries:
```
brew install python
```

Here are a couple other solutions that are reasonable to recommend if a user doesn't want to use Homebrew's Python for some reason: https://stackoverflow.com/questions/70729330/python-on-m1-mbp-trying-to-connect-to-usb-devices-nobackenderror-no-backend-a/70730920#70730920

This suggests that setting `DYLD_FALLBACK_LIBRARY_PATH` should be preferred over `DYLD_LIBRARY_PATH`: https://github.com/orgs/Homebrew/discussions/3424#discussioncomment-3131452

I think we should catch `NoBackendError` and provide the user with a more informative error message.

We may also want to consider updating the documentation with a way to ensure that libusb can be found by pyusb. This could be as simple as doing `brew install python libusb` instead of just `brew install libusb`.

We may even want to consider a workaround in software such as having `cynthion setup` configure `DYLD_FALLBACK_LIBRARY_PATH` or #gasp# look for `/opt/homebrew/lib` specifically: https://stackoverflow.com/questions/70261178/pyusb-no-backend-available-how-to-specify-custom-location-of-libusb

ConceitedCode, 2024-07-16:
Ran into this issue on Ubuntu 24.04 LTS. 

I was able to resolve this by running `pip3 install pyfwup` and flashing the experimental firmware provided in #138.

I confirmed libusb-1.0 package was already installed when the issue occurred.

The immediate issue is resolved for me, but figured I'd document what worked for me in case someone else runs into this issue. 🙂

pktmterenceg, 2024-07-22:
Thanks for everyone's efforts here. 

I've tried the `brew install python libusb` approach, but it did not appear to make a difference. I also tried `pip3 install pyfwup`, just to avoid wasting everyone's time, but got back "Requirement already satisfied."

depili, 2024-07-22:
if your pip isn't complaining and refusing to install anything due to being externally managed if run outside of a venv you aren't using the homebrew pip and python.

pktmterenceg, 2024-07-23:
Thanks - here's the full output: 

`$ pip3 install pyfwup
Requirement already satisfied: pyfwup in /Users/user/miniforge3/lib/python3.10/site-packages (0.4.0)
Requirement already satisfied: pyusb in /Users/user/miniforge3/lib/python3.10/site-packages (from pyfwup) (1.2.1)
Requirement already satisfied: tqdm in /Users/user/miniforge3/lib/python3.10/site-packages (from pyfwup) (4.66.1)`

k3ns, 2025-02-03:
If you are using python in a conda environment, activate the environment, then
conda install conda-forge::libusb