greatscottgadgets/cynthion #140: sudo: cynthion: command not found

mossmann, 2024-07-14:
After installing in a venv or with `pip install --user`, the `cynthion` executable is not typically in the root user's path. This undermines the instruction to run `sudo cynthion setup` to install udev rules.
```
(venv) mossmann@costard:~/src/cynthion/cynthion/python$ cynthion setup
Installing: Linux udev rules
❌      Root privileges are required for installation

Please run 'sudo cynthion setup' to install.
(venv) mossmann@costard:~/src/cynthion/cynthion/python$ sudo cynthion setup
[sudo] password for mossmann: 
sudo: cynthion: command not found
```
Workaround: Follow the alternative instructions to install udev rules manually: https://cynthion.readthedocs.io/en/latest/getting_started.html#install-udev-rules-linux-only

mossmann, 2024-07-15:
There can be similar issues with `PYTHONPATH` and other Python configuration. It looks like the best solution may be to have `cynthion setup` call, for example, `sudo cp` instead of having `sudo cynthion setup` call `cp`.

mossmann, 2024-07-15:
In my case I was able to make `sudo cynthion` work by commenting out the `Defaults secure_path` line and also adding `Defaults:%sudo env_keep += "PYTHONPATH"` in `/etc/sudoers`, but this is not a configuration I think we should generally recommend.

martinling, 2024-07-15:
> It looks like the best solution may be, for example, to have `cynthion setup` call `sudo cp` instead of having `sudo cynthion setup` call `cp`.

That sounds like the right approach to me.

PR #146 (2024-07-16): cli: Call sudo if needed in cynthion setup command.
Fixes #140.
closed 2024-07-16 by antoinevg