greatscottgadgets/cynthion #144: Facedancer + usb_proxy.py(example code) fails to forward Wireless Mouse (12Mbps) to Target

villu164, 2024-07-15:
I'm guessing it might be the HUB (see dmesg from below), I'll see if I can find some USB C device, that doesn't need the HUB

```py
#!/usr/bin/env python3
#
# This file is part of Facedancer.
#
""" USB Proxy example; forwards all USB transactions and logs them to the console. """

from facedancer          import main

from facedancer.proxy    import USBProxyDevice
from facedancer.filters  import USBProxySetupFilters, USBProxyPrettyPrintFilter

# replace with the proxied device's information
ID_VENDOR  = 0x0250
ID_PRODUCT = 0x3412

if __name__ == "__main__":
    # create a USB Proxy Device
    proxy = USBProxyDevice(idVendor=ID_VENDOR, idProduct=ID_PRODUCT)

    # add a filter to forward control transfers between the target host and
    # proxied device
    proxy.add_filter(USBProxySetupFilters(proxy, verbose=0))

    # add a filter to log USB transactions to the console
    proxy.add_filter(USBProxyPrettyPrintFilter(verbose=5))

    main(proxy)
```
throws exception

```sh
(.venv) ➜  luna git:(main) ✗ python usbproxy.py
Traceback (most recent call last):
  File "usbproxy.py", line 18, in <module>
    proxy = USBProxyDevice(idVendor=ID_VENDOR, idProduct=ID_PRODUCT)
  File "/Users/villuorav/git/luna/.venv/lib/python3.8/site-packages/facedancer/proxy.py", line 51, in __init__
    device_handle = self.proxied_device.open(device, detach=True)
  File "/Users/villuorav/git/luna/.venv/lib/python3.8/site-packages/facedancer/proxy.py", line 377, in open
    cls.device_handle.detachKernelDriver(number)
  File "/Users/villuorav/git/luna/.venv/lib/python3.8/site-packages/usb1/__init__.py", line 1199, in detachKernelDriver
    mayRaiseUSBError(
  File "/Users/villuorav/git/luna/.venv/lib/python3.8/site-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/Users/villuorav/git/luna/.venv/lib/python3.8/site-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorAccess: LIBUSB_ERROR_ACCESS [-3]
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/Users/villuorav/git/luna/.venv/lib/python3.8/site-packages/facedancer/proxy.py", line 357, in _destroy_libusb_context
    cls.device_handle.releaseInterface(number)
  File "/Users/villuorav/git/luna/.venv/lib/python3.8/site-packages/usb1/__init__.py", line 1155, in releaseInterface
    mayRaiseUSBError(
  File "/Users/villuorav/git/luna/.venv/lib/python3.8/site-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/Users/villuorav/git/luna/.venv/lib/python3.8/site-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorNotFound: LIBUSB_ERROR_NOT_FOUND [-5]
```

I can visually see, that the python script is able to do something to the mouse connection

https://github.com/user-attachments/assets/d9ee0a18-28cc-4098-a0a2-0da9d75e7dfa

And also from the dmesg, I see some activity

```sh
[4810048.034732]: python3.8@: AppleUSBHostUserClient::openGated: failed to open USB2.0 Hub@01100000: provider is already opened for exclusive access by AppleUSB20Hub
[... 76 lines cut ...]
[4810048.266147]: LQM-WIFI-CT: DPS Symptoms ARP:0 DNSTot:1 DNSImp:1 UsrImp:0 StallScore:50 AppScore:1065353216 NetScore:25
[4810048.266163]: com.apple.p2p: DPS Symptoms ARP:0 DNSTot:1 DNSImp:1 UsrImp:0 StallScore:50 AppScore:1065353216 NetScore:25 RSSI:-52 CCA:6 AWDL:1
[4810048.266168]: com.apple.p2p: DPS Symptoms trigger dampened due to quota
[4810048.356134]: memorystatus_update(enter): pid 77311, priority 210, dirty=0x0, Active(-1MB NF), Inactive(-1MB, NF)
[4810048.356164]: memorystatus_update: init: limit on pid 77311 (-1MB F ) targeting priority(210) dirty?=0x0 
[4810048.373967]: memorystatus_update(enter): pid 77311, priority 40, dirty=0x0, Active(50MB NF), Inactive(50MB, NF)
[4810048.373988]: memorystatus_update: init: limit on pid 77311 (50MB NF) targeting priority(40) dirty?=0x0 
[4810048.375322]: 69 duplicate reports for Sandbox: imagent(7355) deny(1) user-preference-write com.apple.messages.commsafetycom.apple.p2p.stats: AWDL-CCA: channel 36 CCA:0%[S:0 O:0 I:0] Avg-CCA:0%[S:0 O:0 I:0] 
[4810048.499100]: com.apple.p2p: isInfraRealtimePacketThresholdAllowed allowed:0 option:32 threshold:50 noRegistrations:1 cachedPeerCount:1 fastDiscoveryInactive:0 fastDiscoveryOnSince:59082
[4810048.499124]: com.apple.p2p: setScheduleState[9656]: reason:RptDataPathThres sc:Data and force:NO, AWDL-restore:No
[4810048.499243]: 4810048.499242 wlan0.A[12953287] [ik] addDictionaryChannelReasonForJoining@3401:Out of range reasonForJoining 268435456
[4810048.499281]: 4810048.499280 wlan0.A[12953288] [ik] addDictionaryChannelReasonForLeaving@3586:Out of range reasonForLeaving 268435456
[4810048.499385]: 4810048.499385 wlan0.A[12953289] [ik] addDictionaryChannelReasonForJoining@3401:Out of range reasonForJoining 268468224
[4810048.499397]: 4810048.499397 wlan0.A[12953290] [ik] addDictionaryChannelReasonForLeaving@3586:Out of range reasonForLeaving 268468224
[4810048.499454]: 4810048.499454 wlan0.A[12953291] [ik] addDictionaryChannelReasonForJoining@3401:Out of range reasonForJoining 268435456
[4810048.499466]: 4810048.499466 wlan0.A[12953292] [ik] addDictionaryChannelReasonForLeaving@3586:Out of range reasonForLeaving 268435456
[4810048.913141]: com.apple.p2p: monitorAWDLState[8889] : Active Sockets true ValidSvc 2 NumAirplay 0
[4810048.913162]: com.apple.p2p: monitorAWDLState[8972]: BonJourTrig 1 ValidSvc 2 RTApp 0 TSReq 0 HasActAirDrop 0 SocketsActive 1 
[4810048.913166]: com.apple.p2p: setScheduleState[9656]: reason:unknown sc:Data and force:NO, AWDL-restore:No
[4810048.913194]: com.apple.wifi: LQM-WiFi:AWDL State #15 Low Power(4)  DutyCycle 34  StateDuration 5lu.105s[2230058.654 - 2230063.760]   StateComplete  1
[4810048.913197]: com.apple.wifi: LQM-WiFi:AWDL State #16 Data(5)  DutyCycle 76  StateDuration 0lu.941s[2230063.760 - 0.000]   StateComplete  0
[4810048.913201]: com.apple.wifi: LQM-WiFi:AWDL Active Time 2107ms(41%) 5.039s[2230059.662s - 2230064.701s]
[4810048.916139]: LQM-WIFI: WeightAvgLQM rssi=-51 snr=21 txRate=614250 rxRate=656500

```

villu164, 2024-07-15:
Also tried with,
```
Bus 001 Device 002: ID 0781:5596 SanDisk Corporation Ultra T C  Serial: 4C530001110718113092
```
still fails, but for some reason I see Google Chrome having some fingers there

Added some dmesg-s

[4811907_thumb1_chrome_closed.txt](https://github.com/user-attachments/files/16240525/4811907_thumb1_chrome_closed.txt)
[4812031_thumb2_chrome_closed.txt](https://github.com/user-attachments/files/16240526/4812031_thumb2_chrome_closed.txt)
[4812142_thumb3_chrome_open.txt](https://github.com/user-attachments/files/16240528/4812142_thumb3_chrome_open.txt)

antoinevg, 2024-07-25:
Thank you for the report, it sounds like the Cynthion SoC firmware is losing the connection to the proxied device.

Will follow up with more questions if I can't reproduce it on my side!

dominikfehr, 2024-08-05:
same issue with multiple devices.

LIBUSB_ERROR_NOT_FOUND as user
LIBUSB_ERROR_BUSY with sudo privileges

mossmann, 2024-08-10:
I've implemented a similar solution in https://github.com/greatscottgadgets/facedancer/pull/104 which has a couple advantages over the above patch.

tomikoski, 2024-08-15:
If someone else is struggling still with this issue, following worked for me (in ARM macOS and Intel Debian)

* Create build from sources (see guide https://github.com/greatscottgadgets/cynthion/blob/main/BUILD.md)
* Install facedancer from sources (see guide https://facedancer.readthedocs.io/en/latest/getting_started.html)
* Flash build from `firmware` using `cargo run --release`, do notice that this will fail during last step but doesn't matter (due to reset)

If all above works, this will work now:
```
cd cynthion.git/firmware/moondancer
./scripts/moondancer-info.py

# outputs something like:
Found a Cynthion in Moondancer mode!
  Board ID: 16
  Firmware version: r1.0
  Part ID: a0000...
  Serial number: <REDACTED>
```

Exactly the same happens for me in macOS and Debian. With this approach everything works until Cynthion is reseted (or rebooted). If this happens, you just need to rerun flashing part:
```
cargo run --release
```

Once again, good way to test it before use:
```
cd cynthion.git/firmware/moondancer
./scripts/moondancer-info.py
```

tomikoski, 2024-08-15:
One more finding; if I do a `apollo flash-info`, I will have to reflash Cynthion again to make it work. It seems that this part

https://github.com/greatscottgadgets/cynthion/blob/0f635266b282d6cb1f59045a28e3b0b3a65cdadd/firmware/moondancer/src/bin/moondancer.rs#L358-L360

causes it to happen. This was found by monitoring UART.

antoinevg, 2024-08-16:
> One more finding; if I do a `apollo flash-info`, I will have to reflash Cynthion again to make it work.

That's correct. The FPGA shares the USB port with the microcontroller running Apollo. This makes it possible to handover control from the FPGA to Apollo but the microcontroller has no corresponding mechanism to return control to the FPGA.

If you do a `cargo flash facedancer` you can always get back by just pressing the `RESET` button without having to reload the Facedancer bitstream.

patois, 2024-10-10:
I'm not sure whether the underlying problem is a related one but I'm having similar issues with usbproxy.py, no matter the type of USB device I'm trying to proxy (tried multiple USB mass storage devices, mice, keyboards with no success). For some reason, libusb is unable to open the device (I've triple checked device and vendor IDs).

```bash
uname -a
Linux xyz 6.6.47+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux

cynthion info
Cynthion version: 0.1.7
Apollo version: 1.1.0
Python version: 3.11.2 (main, Aug 26 2024, 07:20:54) [GCC 12.2.0]

Found Apollo stub interface!
        Bitstream: Facedancer (Cynthion Project)
        Vendor ID: 1d50
        Product ID: 615b
        bcdDevice: 0104
        Bitstream serial number: xxxxxxxxxxxxxxxx
```

```bash
 ./Facedancer/examples/usbproxy.py
Traceback (most recent call last):
  File ".venv/cynthion/./Facedancer/examples/usbproxy.py", line 21, in <module>
    proxy = USBProxyDevice(idVendor=ID_VENDOR, idProduct=ID_PRODUCT)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".venv/cynthion/lib/python3.11/site-packages/facedancer/proxy.py", line 53, in __init__
    device_handle = self.proxied_device.open(device, detach=True)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".venv/cynthion/lib/python3.11/site-packages/facedancer/proxy.py", line 374, in open
    cls.device_handle = device.open()
                        ^^^^^^^^^^^^^
  File ".venv/cynthion/lib/python3.11/site-packages/usb1/__init__.py", line 2055, in open
    mayRaiseUSBError(libusb1.libusb_open(self.device_p, byref(handle)))
  File ".venv/cynthion/lib/python3.11/site-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File ".venv/cynthion/lib/python3.11/site-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorAccess: LIBUSB_ERROR_ACCESS [-3]

```

any help is greatly appreciated!

antoinevg, 2024-10-11:
If you're on Linux the issue may be that the device you are trying to proxy has not been set up for access by a non-root account.

You can test this quickly by trying to run the USB Proxy script as root:

`sudo ./Facedancer/examples/usbproxy.py`

If this solves it you can set the proxied device up for non-root access by adding a rule to your system's `/etc/udev/rules.d` directory.

You can use the custom rules we install for Cynthion as a reference: [54-cynthion.rules](https://raw.githubusercontent.com/greatscottgadgets/cynthion/main/cynthion/python/assets/54-cynthion.rules)

Basically you'll want to add a line such as:

```
SUBSYSTEM=="usb", ATTR{idVendor}=="replace with proxied device vendorId", ATTR{idProduct}=="replace with proxied device productId", TAG+="uaccess"
```

More info: https://wiki.debian.org/udev

As far as I know macOS doesn't have a similar mechanism so you will probably need to run USB Proxy as root if you're on a Mac.

patois, 2024-10-11:
> You can test this quickly by trying to run the USB Proxy script as root:
> 
> `sudo ./Facedancer/examples/usbproxy.py`
> 

Thank you very much, that fixes my issue indeed (had to become root and activate my Python venv first).

Yet, forwarding USB thumb drives or a keyboard to a Windows host all failed (Windows would not recognize any of them properly).

antoinevg, 2024-10-17:
That could be the problem, we've got some issues with Low Speed support that should be fixed in the next Facedancer release.

What are your experiences with Full and High speed devices?

patois, 2024-10-24:
> That could be the problem, we've got some issues with Low Speed support that should be fixed in the next Facedancer release.
> 
> What are your experiences with Full and High speed devices?

The USB drives I have been testing with are properly being detected by Facedancer as HIGH speed devices but proxying them from a Raspberry Pi 3b+ to a Windows 11 machine still does not work in that they won't be properly recognized/mounted on the target system. There are no obvious errors shown by Facedancer, its logs look clean.

patois, 2025-03-31:
As of today, I still didn't manage to proxy *any* device, no matter the target OS. Updated Cynthion, facedancer, tried different cables, no luck. Proxying a keyboard kind of worked but there's still something wrong because it is stuck sending a single key repeatedly although it isn't held down. Any ideas?