greatscottgadgets/cynthion #151: No backend available (macOS m3)

Oliver0804, 2024-07-22:
I use conda as my Python environment and use Python 3.11 version.
I installed libusb via brew install libusb.
However, I still get the following information:

```

Traceback (most recent call last):
  File "/Users/oliver/anaconda3/envs/luan/bin/cynthion", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/oliver/anaconda3/envs/luan/lib/python3.11/site-packages/cynthion/commands/cli.py", line 91, in main
    bitstream_info = get_bitstream_information()
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/oliver/anaconda3/envs/luan/lib/python3.11/site-packages/cynthion/commands/util.py", line 30, in get_bitstream_information
    device = ApolloDebugger._find_device([
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/oliver/anaconda3/envs/luan/lib/python3.11/site-packages/apollo_fpga/__init__.py", line 172, in _find_device
    device = usb.core.find(backend=backend, idVendor=vid, idProduct=pid, custom_match=custom_match)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/oliver/anaconda3/envs/luan/lib/python3.11/site-packages/usb/core.py", line 1309, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend availabl

```

depili, 2024-07-22:
Duplicate of https://github.com/greatscottgadgets/cynthion/issues/136. Non-homebrew pythons aren't patched to find the homebrew libraries.

mossmann, 2024-07-22:
@Oliver0804 Does `brew install python` or one of the other solutions mentioned in #136 work for you?

Oliver0804, 2024-07-23:
> @Oliver0804 Does `brew install python` or one of the other solutions mentioned in #136 work for you?

I prefer not to use brew install python to set up the Python environment. In fact, I'm more accustomed to using conda for creating environments, given the numerous versions of Python and the significant dependency issues.

My solution is to use brew install libusb and then add the following to ~/.zshenv:

```sh
export DYLD_LIBRARY_PATH="/opt/homebrew/opt/libusb/lib:$DYLD_LIBRARY_PATH"
```
This approach temporarily resolves the issue.

closed 2024-07-23 by Oliver0804