greatscottgadgets/cynthion #153: Execution of update command failed under Arch Linux

qrvstw, 2024-07-23:
I have installed on my Arch Linux the cynthion version 0.1.2. During the
execution of update coomand I get errors. Here is the output from my terminal:
[user]~> cynthion setup —check
Checking: Linux udev rules
✅ 54-cynthion.rules is present
✅ 54-cynthion.rules matches factory rules
All checks completed successfully.
[user]~> cynthion info
Detected a Cynthion device!
Hardware: Cynthion r1.4
Serial number: ELGCYXKMGJJVCIBAEAMDCJAOQN
Firmware version: v1.0.1
USB API version: 1.0
Flash UID: 0000000000000000
[user]~> cynthion update
Updating FPGA configuration flash with 253603 bytes…
Operation complete!
Updating device firmware with 13816 bytes…
Traceback (most recent call last):
File "/usr/bin/cynthion", line 8, in <module>
sys.exit(main())
^^^^^^
File "/usr/lib/python3.12/site-packages/cynthion/commands/cli.py", line 112, in
main
args.func(device, args)
File "/usr/lib/python3.12/site-packages/cynthion/commands/cynthion_update.py",
line 21, in cynthion_update
flash_mcu_firmware(device, find_cynthion_asset("apollo.bin"))
File "/usr/lib/python3.12/site-packages/cynthion/commands/util.py", line 134,
in flash_mcu_firmware
board.program(firmware, status_callback=lambda written, _:
progress.update(written))
File "/usr/lib/python3.12/site-packages/fwup/dfu.py", line 278, in program
self.__raw_write_page(page_address, data_to_program)
File "/usr/lib/python3.12/site-packages/fwup/dfu.py", line 248, in
__raw_write_page
self.__dfu_out_request(self.DFU_DOWNLOAD, block_number, data)
File "/usr/lib/python3.12/site-packages/fwup/dfu.py", line 191, in
__dfu_out_request
self.device.ctrl_transfer(self.USB_CLASS_OUT_REQUEST_TO_INTERFACE, request,
value,
File "/usr/lib/python3.12/site-packages/usb/core.py", line 1080, in
ctrl_transfer
self._ctx.managed_claim_interface(self, interface_number)
File "/usr/lib/python3.12/site-packages/usb/core.py", line 113, in wrapper
return f(self, args, *kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3.12/site-packages/usb/core.py", line 178, in
managed_claim_interface
self.backend.claim_interface(self.handle, i)
File "/usr/lib/python3.12/site-packages/usb/backend/libusb1.py", line 829, in
claim_interface
_check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
File "/usr/lib/python3.12/site-packages/usb/backend/libusb1.py", line 604, in
_check
raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 2] Entity not found

mossmann, 2024-07-23:
Can you please try: `cynthion update --mcu-firmware`

mossmann, 2024-07-25:
Try pressing PROGRAM. This should cause LEDs B-D to turn off. Then try:
```
fwup-util --device 1d50:615c --info
```
What output does that give you?
Does LED C start flashing?

mossmann, 2024-07-25:
@qrvstw Do you happen to be using Arch on top of Windows Subsystem for Linux?

qrvstw, 2024-07-25:
[user]~> fwup-util --device 1d50:615c --info

Target found!
[user] ->
The LED C does not flashing.

qrvstw, 2024-07-25:
"Do you happen to be using Arch on top of Windows Subsystem for Linux?"

No, it is a native installation.

Today I have updated the following packages and try the update again:
python-apollo to 1.1.0
python-pyfwup to 0.5.0
cynthion-firmware-bin to 0.1.3
python-cynthon to 0.1.3

[user]~> cynthion update
Updating FPGA configuration flash with 258874 bytes...
Operation complete!
Updating device firmware with 13876 bytes...
Traceback (most recent call last):                                              
  File "/usr/bin/cynthion", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.12/site-packages/cynthion/commands/cli.py", line 94, in main
    args.func(device, args)
  File "/usr/lib/python3.12/site-packages/cynthion/commands/cynthion_update.py", line 21, in cynthion_update
    flash_mcu_firmware(device, find_cynthion_asset("apollo.bin"))
  File "/usr/lib/python3.12/site-packages/cynthion/commands/util.py", line 105, in flash_mcu_firmware
    board.program(firmware, status_callback=lambda written, _: progress.update(written))
  File "/usr/lib/python3.12/site-packages/fwup/dfu.py", line 292, in program
    self.__raw_write_page(page_address, data_to_program)
  File "/usr/lib/python3.12/site-packages/fwup/dfu.py", line 262, in __raw_write_page
    self.__dfu_out_request(self.DFU_DOWNLOAD, block_number, data)
  File "/usr/lib/python3.12/site-packages/fwup/dfu.py", line 205, in __dfu_out_request
    self.device.ctrl_transfer(self.USB_CLASS_OUT_REQUEST_TO_INTERFACE, request, value,
  File "/usr/lib/python3.12/site-packages/usb/core.py", line 1080, in ctrl_transfer
    self._ctx.managed_claim_interface(self, interface_number)
  File "/usr/lib/python3.12/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/usb/core.py", line 178, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python3.12/site-packages/usb/backend/libusb1.py", line 829, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python3.12/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 2] Entity not found

mossmann, 2024-07-28:
Can you show us what is reported by `dmesg` around the time of the `Entity not found` error?

qrvstw, 2024-07-28:
It works now.  The update process was successfully and said "Operation complete!".
The usbguard service is running on my system and it blocks all unknown USB devices.  During the update process, another USB device is activated, which I had not seen.  After I had added both USB devices to the list of permanently allowed USB devices, the update also works.
Did I overlook the two USB devices in the documentation?
If not, please add this important information to the documentation.

martinling, 2024-07-28:
During the update process you will see three different devices appear:
- The Cynthion analyzer gateware running on the FPGA (`Cynthion USB Analyzer`, `1d50:615b`)
- The Apollo debugger running on the MCU (`Cynthion Apollo Debugger`, `1d50:615c`)
- The Saturn-V bootloader running on the MCU (`Cynthion Bootloader`, `1d50:615c`)

straithe, 2024-10-17:
I'm closing this as there hasn't been a response in over 30 days. Please re-open this issue or open a new one if you still need assistance.