greatscottgadgets/cynthion #158: Cynthion update fails Windows 10

iBringsTheRuckus, 2024-08-03:
I'm currently running an updated version of Windows 10.
Python 3.12.4

I run `cynthion info --force-offline` and everything is normal.
I ran `pip install --upgrade cynthion` and went well.

I then run `cynthion update` and receive the following error. After the error occurs, the C LED is continuously flashing.
`Updating FPGA configuration flash with 258874 bytes...
Operation complete!
Updating device firmware with 13876 bytes...
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Scripts\cynthion.exe\__main__.py", line 7, in <module>
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\cynthion\commands\cli.py", line 94, in main
    args.func(device, args)
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\cynthion\commands\cynthion_update.py", line 21, in cynthion_update
    flash_mcu_firmware(device, find_cynthion_asset("apollo.bin"))
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\cynthion\commands\util.py", line 105, in flash_mcu_firmware
    board.program(firmware, status_callback=lambda written, _: progress.update(written))
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\fwup\dfu.py", line 292, in program
    self.__raw_write_page(page_address, data_to_program)
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\fwup\dfu.py", line 262, in __raw_write_page
    self.__dfu_out_request(self.DFU_DOWNLOAD, block_number, data)
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\fwup\dfu.py", line 205, in __dfu_out_request
    self.device.ctrl_transfer(self.USB_CLASS_OUT_REQUEST_TO_INTERFACE, request, value,
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\core.py", line 1071, in ctrl_transfer
    self._ctx.managed_open()
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\core.py", line 131, in managed_open
    self.handle = self.backend.open_device(self.dev)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\backend\libusb1.py", line 804, in open_device
    return _DeviceHandle(dev)
           ^^^^^^^^^^^^^^^^^^
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\backend\libusb1.py", line 652, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\backend\libusb1.py", line 600, in _check
    raise NotImplementedError(_strerror(ret))
NotImplementedError: Operation not supported or unimplemented on this platform
Traceback (most recent call last):
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\weakref.py", line 666, in _exitfunc
    f()
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\weakref.py", line 590, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\_objfinalizer.py", line 106, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\_objfinalizer.py", line 73, in _do_finalize_object
    self._finalize_object()
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\backend\libusb1.py", line 613, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000000`

mossmann, 2024-08-03:
Please press RESET and then try:
```
pip install --upgrade pyfwup
cynthion update --mcu-firmware
```

iBringsTheRuckus, 2024-08-03:
`C:\Users\Admin\Documents>pip install --upgrade pyfwup
Requirement already satisfied: pyfwup in c:\users\admin\appdata\local\programs\python\python312\lib\site-packages (0.5.1)
Requirement already satisfied: pyusb in c:\users\admin\appdata\local\programs\python\python312\lib\site-packages (from pyfwup) (1.2.1)
Requirement already satisfied: tqdm in c:\users\admin\appdata\local\programs\python\python312\lib\site-packages (from pyfwup) (4.66.4)
Requirement already satisfied: colorama in c:\users\admin\appdata\local\programs\python\python312\lib\site-packages (from tqdm->pyfwup) (0.4.6)

C:\Users\Admin\Documents>cynthion update --mcu-firmware
Updating device firmware with 13876 bytes...
Operation complete!
Traceback (most recent call last):
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\weakref.py", line 666, in _exitfunc
    f()
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\weakref.py", line 590, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\_objfinalizer.py", line 106, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\_objfinalizer.py", line 73, in _do_finalize_object
    self._finalize_object()
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\backend\libusb1.py", line 613, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000000`

iBringsTheRuckus, 2024-08-03:
I hit the reset button and then did `cynthion update --mcu-firmware
Updating device firmware with 13876 bytes...
Operation complete!` Which is great!

I then did `cynthion update
Updating FPGA configuration flash with 258874 bytes...
Operation complete!
Updating device firmware with 13876 bytes...
Operation complete!`

And it worked that time! No clue as to why it worked...

mossmann, 2024-08-03:
I suspect that the error had to do with another application or driver trying to talk to the device, so it may be that unpredictable conditions of other software changed.

mossmann, 2024-08-08:
I'm closing this since the problem went away, and I suspect that https://github.com/greatscottgadgets/pyfwup/pull/13 will prevent it from coming back.

closed 2024-08-08 by mossmann