greatscottgadgets/cynthion #161: Cynthion setup needs an attached Cynthion

straithe, 2024-08-04:
I tried to run "cynthion setup" in a command line on a new system after running "pip install cynthion". I got the following error:

```
sraithe@guardGomazoa:~$ cynthion setup
Traceback (most recent call last):
  File "/home/straithe/.local/bin/cynthion", line 8, in <module>
    sys.exit(main())
  File "/home/straithe/.local/lib/python3.10/site-packages/cynthion/commands/cli.py", line 93, in main
    device = ApolloDebugger(force_offline=force_offline)
  File "/home/straithe/.local/lib/python3.10/site-packages/apollo_fpga/__init__.py", line 110, in __init__
    raise DebuggerNotFound("No Apollo debugger or stub interface found.")
apollo_fpga.DebuggerNotFound: No Apollo debugger or stub interface found.
```

I am guessing it is because I did not have a Cynthion plugged into my computer while running this command.

straithe, 2024-08-05:
I plugged a Cynthion into my system and got this error:

```
cynthion setup
Traceback (most recent call last):
  File "/home/straithe/.local/lib/python3.10/site-packages/apollo_fpga/__init__.py", line 116, in __init__
    self._request_handoff(fpga_device)
  File "/home/straithe/.local/lib/python3.10/site-packages/apollo_fpga/__init__.py", line 155, in _request_handoff
    device.ctrl_transfer(request_type, REQUEST_APOLLO_ADV_STOP, wIndex=intf_number, timeout=5000)
  File "/home/straithe/.local/lib/python3.10/site-packages/usb/core.py", line 1071, in ctrl_transfer
    self._ctx.managed_open()
  File "/home/straithe/.local/lib/python3.10/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/home/straithe/.local/lib/python3.10/site-packages/usb/core.py", line 131, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/home/straithe/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 804, in open_device
    return _DeviceHandle(dev)
  File "/home/straithe/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 652, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/home/straithe/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/straithe/.local/bin/cynthion", line 8, in <module>
    sys.exit(main())
  File "/home/straithe/.local/lib/python3.10/site-packages/cynthion/commands/cli.py", line 93, in main
    device = ApolloDebugger(force_offline=force_offline)
  File "/home/straithe/.local/lib/python3.10/site-packages/apollo_fpga/__init__.py", line 118, in __init__
    raise DebuggerNotFound(f"Handoff request failed: {e.strerror}")
apollo_fpga.DebuggerNotFound: Handoff request failed: Access denied (insufficient permissions)
```

straithe, 2024-08-05:
Manually running the udev rules listed on https://cynthion.readthedocs.io/en/latest/getting_started.html fixed the second error I experienced.

litneet64, 2024-08-05:
I had the same problem when trying to run `cynthion setup` as a regular unprivileged user. Afterwards I ran it with sudo and it worked fine (I also installed cynthion as a globally available python package first).

PR #164 (2024-08-06): cli: Don't require device to create udev rule
Fix #161
closed 2024-08-06 by antoinevg