greatscottgadgets/cynthion #165: Errno 19 while updating Cyntion

w1jp, 2024-08-05:
I am new to the Cynthion just having received mine this last week. I am following the [getting started guide](https://cynthion.readthedocs.io/en/latest/getting_started.html#updating-cynthion-microcontroller-firmware-and-fpga-configuration-flash) and when it comes to updating the firmware, I get this error:
```
(pye) cynthion % cynthion update
Updating FPGA configuration flash with 258874 bytes...
Operation complete!
Traceback (most recent call last):
  File "/Users/jon/wrk/cynthion/pye/bin/cynthion", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/jon/wrk/cynthion/pye/lib/python3.12/site-packages/cynthion/commands/cli.py", line 94, in main
    args.func(device, args)
  File "/Users/jon/wrk/cynthion/pye/lib/python3.12/site-packages/cynthion/commands/cynthion_update.py", line 21, in cynthion_update
    flash_mcu_firmware(device, find_cynthion_asset("apollo.bin"))
  File "/Users/jon/wrk/cynthion/pye/lib/python3.12/site-packages/cynthion/commands/util.py", line 98, in flash_mcu_firmware
    board = DFUTarget(idVendor=shared.usb.bVendorId.apollo, idProduct=shared.usb.bProductId.apollo)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jon/wrk/cynthion/pye/lib/python3.12/site-packages/fwup/dfu.py", line 172, in __init__
    self.__init__(index=index, detach=False, *args, **kwargs)
  File "/Users/jon/wrk/cynthion/pye/lib/python3.12/site-packages/fwup/dfu.py", line 158, in __init__
    self.__read_device_info()
  File "/Users/jon/wrk/cynthion/pye/lib/python3.12/site-packages/fwup/dfu.py", line 185, in __read_device_info
    for configuration in self.device:
  File "/Users/jon/wrk/cynthion/pye/lib/python3.12/site-packages/usb/core.py", line 1140, in __iter__
    yield Configuration(self, i)
          ^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jon/wrk/cynthion/pye/lib/python3.12/site-packages/usb/core.py", line 605, in __init__
    desc = backend.get_configuration_descriptor(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jon/wrk/cynthion/pye/lib/python3.12/site-packages/usb/backend/libusb1.py", line 773, in get_configuration_descriptor
    _check(self.lib.libusb_get_config_descriptor(
  File "/Users/jon/wrk/cynthion/pye/lib/python3.12/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 19] No such device (it may have been disconnected)
```

Note also that the *C LED* is flashing on the device.

w1jp, 2024-08-05:
UPDATE: I tried `--bitstream` which succeded; however `--mcu-firmware` provided the same error.

mossmann, 2024-08-05:
Can you please try `cynthion update --mcu-firmware` after updating pyfwup with this PR? https://github.com/greatscottgadgets/pyfwup/pull/13

If you need help installing from the PR, let me know.

w1jp, 2024-08-07:
Thank you-- that did the trick. 
```
Cynthion version: 0.1.3
Apollo version: 1.1.0
Python version: 3.12.4 (main, Jun  6 2024, 18:26:44) [Clang 15.0.0 (clang-1500.3.9.4)]

Found Apollo stub interface!
	Bitstream: USB Analyzer (Cynthion Project)
	Vendor ID: 1d50
	Product ID: 615b
	bcdDevice: 0104
	Bitstream serial number: 362c84cba3ac60de

Forcing offline.

Found Cynthion device!
	Hardware: Cynthion r1.4
	Manufacturer: Great Scott Gadgets
	Product: Cynthion Apollo Debugger
	Serial number: AWD6G6SMGJJVCIBAEA3EGCIP74
	Vendor ID: 1d50
	Product ID: 615c
	bcdDevice: 0104
	Firmware version: v1.1.0
	USB API version: 1.2
	Flash UID: 362c84cba3ac60de
```

closed 2024-08-07 by w1jp