greatscottgadgets/cynthion #169: "ValueError: The device has no langid" after installing Facedancer on Windows

pdaderko, 2024-08-10:
I recently received my Cynthion device, and have been working through the "Getting Started" guide ( https://cynthion.readthedocs.io/en/latest/getting_started.html ).  Everything was successful through the "Using Cynthion with Packetry" step, but the "Using Cynthion with Facedancer" ( https://cynthion.readthedocs.io/en/latest/getting_started_facedancer.html ) seems to break Cynthion functionality for me.

Everything appears to be successful until the "cynthion info" step, which returns:
```
C:\Users\DogP>cynthion info
Cynthion version: 0.1.3
Apollo version: 1.1.0
Python version: 3.11.1 (tags/v3.11.1:a7a450f, Dec  6 2022, 19:58:39) [MSC v.1934 64 bit (AMD64)]

Found Apollo stub interface!
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\DogP\AppData\Roaming\Python\Python311\Scripts\cynthion.exe\__main__.py", line 7, in <module>
  File "C:\Users\DogP\AppData\Roaming\Python\Python311\site-packages\cynthion\commands\cli.py", line 91, in main
    args.func(args)
  File "C:\Users\DogP\AppData\Roaming\Python\Python311\site-packages\cynthion\commands\cynthion_info.py", line 19, in cynthion_info
    if ApolloDebugger.print_info(force_offline=args.force_offline, out=logging.info,
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\DogP\AppData\Roaming\Python\Python311\site-packages\apollo_fpga\__init__.py", line 436, in print_info
    out(f"\tBitstream: {device.product} ({device.manufacturer})")
                        ^^^^^^^^^^^^^^
  File "C:\Users\DogP\AppData\Roaming\Python\Python311\site-packages\usb\core.py", line 875, in product
    self._product = util.get_string(self, self.iProduct)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\DogP\AppData\Roaming\Python\Python311\site-packages\usb\util.py", line 313, in get_string
    raise ValueError("The device has no langid"
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)
```
And the Facedancer demos fail as well.

If I click the program button, running "cynthion info" is successful again (with only LED A lit).  But I'm no longer able to run analyzer on the PC.  If I run "cynthion run analyzer", LEDs A-D light, and running "cynthion info" outputs the same langid error.  And Packetry now complains "The device is not usable because: Failed to claim interface".

I initially tested on a Windows 10 laptop and got the exact same error.  To make sure the device didn't get flashed a bad image or something, I ran through the process again on a Windows 11 desktop, which again worked great until the Facedancer step.  So it seems to be a reproducible software issue (or I'm consistently doing something wrong).

Lastly, I tried "pip uninstall facedancer" and "pip uninstall cynthion", then starting over from the beginning, but get the langid error from the start.

Any help or ideas?

nekrondev, 2024-08-10:
Did you try to use `Facedancer` and `Packetry` at the same time, i.e. run `Packetry`  with analyzer gateware and then keep `Packetry` open while doing the `cynthion info` command?

I learned that the WinUSB driver allows only bound access to a single process per device. That means if `Packetry` claimed the analyzer device, running `cynthion info` (which is a new process) will get permission problems accessing the analyzer device to grab some infos like the language id. The device had been claimed by `Packetry` process so that process must release the WinUSB driver first before other tools like Cynthion CLI can access it.

You should make sure that only a single process is accessing the device so either Cynthion CLI or Packetry but not both at the same time.

mossmann, 2024-08-10:
We need to update the documentation to say this, but Facedancer only works on Linux and macOS at this time.

pdaderko, 2024-08-10:
> Did you try to use `Facedancer` and `Packetry` at the same time, i.e. run `Packetry` with analyzer gateware and then keep `Packetry` open while doing the `cynthion info` command?
Packetry was closed before trying Facedancer, unless there's a background process keeping something open.

> We need to update the documentation to say this, but Facedancer only works on Linux and macOS at this time.
Ah... that makes sense then!  Would you expect that to break the analyzer functionality on Windows, and are the instructions on completely removing/reinstalling?  I used pip install to install Facedancer but pip uninstall didn't fix it.

Thanks!

pdaderko, 2024-08-10:
It looks like I was able to resolve this by plugging the Cynthion device into the computer, opening Device Manager and manually clicking "Uninstall device" for both "USB Analyzer" devices under "Universal Serial Bus devices"... then right-clicking and selecting Scan for hardware changes to reinstall.  This seems to have brought back the original functionality I had before running Facedancer.

BTW, since I had done this on two different computers, I took before/after screenshots of the 2nd one, and didn't notice any obvious differences between what was installed and what got reinstalled (same driver/version, VID/PID, etc), but presumably there was some minor difference, something cached, etc.

And thanks again for clarification on Facedancer... I'll try it on my Ubuntu laptop.

mossmann, 2024-08-10:
I actually encountered this issue yesterday while trying to test some things, and I was able to resolve it similarly. I wasn't 100% sure at the time that I hadn't done something else wrong to cause the problem, but it matched your experience closely.

These steps worked for me to get the analyzer gateware working again:

- Press PROGRAM
- `cynthion info` for confirmation
- `cynthion flash analyzer`
- Open Device Manager and find pair of USB devices called "USB Analyzer" or "WinUSB Device"
- Right-click and "Uninstall" both devices
- Unplug Cynthion
- Plug in Cynthion
- `cynthion info` for confirmation

closed 2024-08-20 by antoinevg