greatscottgadgets/cynthion #170: Support Facedancer on Windows

mossmann, 2024-08-10:
Warning! Facedancer does not currently work on Windows. Attempts to use Facedancer on Windows [may cause USB analysis to stop working](#169)

There are several things that need to happen to bring Facedancer support to Windows:

- [ ] Fix #169
- [ ] Add WCID descriptors to moondancer firmware. This may be all that is required to fix #169, but we don't know yet.
- [ ] Add WCID descriptors to GreatFET firmware. (This isn't strictly necessary for it to work on Cynthion, but we probably should do it at the same time in order to give us more ways to test things.)
- [ ] Merge https://github.com/greatscottgadgets/facedancer/pull/104
- [ ] Add Windows support to pygreat (https://github.com/greatscottgadgets/libgreat)

Once these things are done, we'll be able to test and find out if there are additional problems. There may be other bugs in Cynthion, Facedancer, or pygreat that affect only Windows.

pickeditmate, 2024-08-17:
Ohhhhhhhh, that explains my predicament then, thought it was because I had previously installed facedancer for greatfet. Was just following readthedocs instructions before defcon and hit exactly this issue. 
Fwiw announcement in discord might be needed too.

antoinevg, 2024-09-19:
GreatFET-related tasks moved to: https://github.com/greatscottgadgets/greatfet/issues/439

closed 2024-09-19 by straithe