greatscottgadgets/cynthion #174: question: Can Cynthion emulate a host?

firelizzard18, 2024-08-14:
I am working on USB support for a microcontroller. When I plug my device into a PC, it does not work. I'm sure there's a bug somewhere in my descriptors or something the host OS doesn't even register that a device has been attached so wireshark is useless. To eliminate variables, I'd like to configure Cynthion to be a really basic USB host so I can plug my device in and capture and analyze the packets it sends without all the extra complexity of a full PC/OS host. Is that supported?

Zman350x, 2024-08-14:
I know this isn't what you're asking, but is it even showing up in device manager at all? If not, then you probably have a bigger issue than device descriptors. [It only takes a few resistors for a device to show up on your PC](https://www.youtube.com/watch?v=VG5bWzEPfsg). Broken descriptors would just give something like "unknown USB device" or "USB error" or something like that.

firelizzard18, 2024-08-15:
I phrased the description as if I was currently working on it but in reality this is something I worked on a few years ago and recently was reminded of. I was using a microcontroller with built-in USB support. The manufacturer of the development board I used provided a C library (Arduino core) that works, but my attempt to convert that to Go failed. So I'm pretty sure I was doing something wrong, not that the microcontroller was bad. But it's been a while.

nekrondev, 2024-08-15:
> I'd like to configure Cynthion to be a really basic USB host so I can plug my device in and capture and analyze the packets it sends without all the extra complexity of a full PC/OS host. Is that supported?

Technically one could write gateware or a SoC firmware that would emulate an USB host. However, currently there is no support for this on the software side as far as I can see. The best for you would be to hook Cynthion up to a PC and capture the traffic between the MCU USB device and PC USB host controller and see whats going on. It's important to select the correct USB speed at `Packetry` before you attach the device or otherwise you won't see any packets.

firelizzard18, 2024-08-15:
> Broken descriptors would just give something like "unknown USB device" or "USB error" or something like that.

This is demonstrably not true at least in Linux. When I program it (with the blinky example) via the manufacturer-provided Arduino core, it shows up in the system log (dmesg) and lsusb. When I program it with a program that has no USB support, nothing shows up in dmesg or lsusb.

miek, 2024-08-15:
> This is demonstrably not true at least in Linux. When I program it (with the blinky example) via the manufacturer-provided Arduino core, it shows up in the system log (dmesg) and lsusb. When I program it with a program that has no USB support, nothing shows up in dmesg or lsusb.
Most microcontrollers allow you to turn the D+/D- pull-up resistor on or off, so what probably happened is that the pull-up was disabled when running the program with no USB support and that's why you don't see anything in `dmesg`

miek, 2024-09-19:
Closing as I think the question has been covered, but feel free to re-open if you have any more questions!

closed 2024-09-19 by miek