greatscottgadgets/cynthion #177: Add brief info to docs about what each use-case is for

x0rloser, 2024-08-20:
When new to cynthion it is not clear what the different uses are for.
I'd submit a pull request for this, but since I am very new to cynthion its probably best someone else writes these.
I suggest adding a brief sentence to the start of the following pages to help make it clear for new users.

docs/source/getting_started_packetry.rst
docs/source/getting_started_facedancer.rst
docs/source/getting_started_usbproxy.rst

For example on the getting_started_packetry.rst page, right at the top under the main heading, you might put a sentence like:
"This uses cynthion as a usb packet capture device between a usb host and target in order to analyze the packets sent."

The facedancer page might have something like:
"This uses cynthion to emulate usb devices"

The usb proxy page might have:
"This uses cynthion to proxy packets between another usb host and a usb target on the control machine."

x0rloser, 2024-08-21:
I would also suggest adding a link to the "Getting started with packetry" doc page at the end of the "Using cynthion with pcketry" doc page, as the cynthion doc page feels like it stops part way through explaining how to use packetry.

So at the bottom of the page:
https://cynthion.readthedocs.io/en/latest/getting_started_packetry.html
(docs/source/getting_started_packetry.rst)

Add a link to one of the following:
https://packetry.readthedocs.io/en/latest/quick_start.html#
https://packetry.readthedocs.io/en/latest/quick_start.html#install-packetry
https://packetry.readthedocs.io/en/latest/quick_start.html#connect-cynthion
https://packetry.readthedocs.io/en/latest/quick_start.html#launch-packetry
https://packetry.readthedocs.io/en/latest/quick_start.html#start-capturing-usb-traffic

Whichever of these links you feel it is best for the reader to get directed to.
The "Connect cynthion" info in the packetry docs is more fully explained than in the cynthion docs, and packetry install is not covered in cynthion docs, and also the steps to follow when capturing traffic is also missing.

Actually I think the install packetry link might be the best choice to link to
https://packetry.readthedocs.io/en/latest/quick_start.html#install-packetry

udif, 2024-09-22:
Here is more information I think is missing:

* Meaning of different LED patterns
* Be more explicit in explaining that Target-A and Target-C are physically shorted
* Explain what AUX USB-C is for
* Explain How the reset/user/program switches are used
* Explain What the A & B signal banks on the side can be used for

* Include a copy of the cable configuration for `packetry --test-cynthion` (I think that for Windows that would be `packetry-cli.exe  --test-cynthion`)
Also mention thyat in Windows low-speed will currently fail with `Test failed: Test device not found`)
https://discord.com/channels/807355811281109082/816773068277809182/1261922256972419143
[image]

straithe, 2024-10-17:
Thank you for the extra documentation suggestions. We will add the extra information when we can.