greatscottgadgets/cynthion #184: I don't think the aux power works as written? How do I send power to a port?

rlewkowicz, 2024-08-30:
[        Resource("aux_vbus_en",        0, Pins("L2", dir="o"), Attrs(IO_TYPE="LVCMOS33")),](https://github.com/greatscottgadgets/cynthion/blob/120007938836513437540b9cc77e5be31f1a2d25/cynthion/python/src/gateware/platform/cynthion_r1_4.py#L156)

Has this been verified to work as written? If yes, whats the code to power the aux port via m.d.comb?

mossmann, 2024-08-31:
`aux_vbus_en` has been tested. The purpose of `aux_vbus_en` when used alone is to pass VBUS power from AUX to TARGET A. If you want to pass power from some other port *to* AUX, you must enable both `aux_vbus_en` and either `target_c_vbus_en` or `control_vbus_en`.

For background on the power pass-through architecture, see: https://www.crowdsupply.com/great-scott-gadgets/cynthion/updates/cynthion-hardware-design-update

One change since that post is that we turned off TARGET C pass-through when Cynthion is unpowered.

Does this answer your question?

rlewkowicz, 2024-08-31:
This is perfect!

closed 2024-08-31 by rlewkowicz

rlewkowicz, 2024-09-01:
[image]
This is a great image just for reference. I know I'm goofy rambling in half these threads, someone will appreciate it one day I promise. I'm learning fpgas through this platform and amaranth.