greatscottgadgets/cynthion #186: Can't get a new Cynthion to work reliably as a bus analyzer

udif, 2024-09-22:
I got a cynthion in the mail a few months ago, and finally had the time to put it into use.
I'm using Windows 11.
I installed cynthion using pip and the latest packetry 0.2.2 from the github releases page.
I also updated the firmware (both MCU and FPGA).
```
PS C:\> cynthion info --force-offline
Cynthion version: 0.1.6
Apollo version: 1.1.0
Python version: 3.11.5 | packaged by Anaconda, Inc. | (main, Sep 11 2023, 13:26:23) [MSC v.1916 64 bit (AMD64)]

Found Apollo stub interface!
        Bitstream: USB Analyzer (Cynthion Project)
        Vendor ID: 1d50
        Product ID: 615b
        bcdDevice: 0104
        Bitstream serial number: xxxxxxxxxxxxxxxx

Forcing offline.

Found Cynthion device!
        Hardware: Cynthion r1.4
        Manufacturer: Great Scott Gadgets
        Product: Cynthion Apollo Debugger
        Serial number: XXXXXXXXXXXXXXXXXXXXXXXXXX
        Vendor ID: 1d50
        Product ID: 615c
        bcdDevice: 0104
        Firmware version: v1.1.0
        USB API version: 1.2
        Flash UID: xxxxxxxxxxxxxxxx
```
(Everything looking like a S/N has been removed ofcourse)

The main issue I'm struggling with, is that a simple USB device that works fine when plugged directly into the target, stops working when plugged through the Target-A and Target-C USB sockets on Cynthion (stops working == not detected correctly by the USB device's proprietary SW).

I made sure everything is working from the HW side:
* The control port is connected to a different Host PC
* The cable from the target PC to cynthion (USB-A to USB-C) is connected to the Target-C USB socket.
* The target device is connected through a USB-A cable to the Target USB-A socket. It has been tested independently by plugging the target device directly into the target PC.
* All cables and cynthion itself have been independently tested by using packetry test mode, as well as testing the cables using other applications (Android file transfer, etc.). I also swapped the cables between the target and host cynthion ports and in both cases cynthion was detected.

Error symptoms:
When starting a capture, the device starts appearing multiple times, with a new instance every 1-2 seconds (and it keeps reproducing):
In addition, the log itself keeps indenting to the right with each new device added.

[image]

(redacted proprietary device name that repeats itself indefinitely)

It looks like something in cynthion is injecting data/noise on the target bus.

martinling, 2024-09-22:
Hi @udif, sorry to hear you're having trouble.

I haven't seen these exact symptoms before, but it seems like what's probably happening is that the signal integrity over the USB connection as a whole is marginal. So the device is able to successfully enumerate at High Speed, but the connection then falters a bit later, at which point the host resets the port and the enumeration starts again, repeating indefinitely.

The log indenting to the right is just a consequence of the repeated enumeration - there is one column created in the connecting lines for each unique endpoint that has traffic in the capture, and each enumeration results in at least one new device and endpoint being used.

You've already been quite thorough in checking that the analyzer and the cables all work individually, and that they work together in the test mode. However, it may still be the case that the cumulative effects of the cables, the Cynthion passthrough, and perhaps also poor receiver performance on the target device and/or host, are collectively just enough to prevent things from working reliably here.

There might also be some particular event that's helping to trigger this - e.g. some spike causing a glitch that occurs as the target device switches something on. Any packets captured just before the re-enumeration may be a clue there.

Unless you are already very confident of the internal structure of the cables you are using, I would suggest looking for some cables that are known to be of the proper internal construction for good signal integrity. At a minimum that means full shielding, properly connected to the connector shells at both ends, and adequate diameter of the power conductors. There are surprisingly many cables on the market that do not meet these basic requirements including some from popular brands, with very attractive braided fabric sleeves and fancy connectors, but dreadful signal integrity.

udif, 2024-09-22:
As the target device has a USB-B socket, I don't have a short USB-A<>USB-B cable to replace it with, but replacing the USB-A<>USB-C cable connected from the target PC to the target USB-C on cynthion with a very short cable (~30cm) seems to solve the problem!

martinling, 2024-09-22:
Great, glad you got it working!

Yes shorter cables are usually better as well - although not universally, as they also make longer antennas for noise.

closed 2024-09-22 by martinling