greatscottgadgets/cynthion #188: Traceback Error: USB Timeout

bluibi5, 2024-09-26:
Posted this on the Discord and was asked to post it here: 

Hello all, I'm getting facedancer set up with the Cynthion and it looks like I was successful: 

```
$ cynthion info
Cynthion version: 0.1.6
Apollo version: 1.1.0
Python version: 3.10.12 (main, Jul 29 2024, 16:56:48) [GCC 11.4.0]

Found Apollo stub interface!
    Bitstream: Facedancer (Cynthion Project)
    Vendor ID: 1d50
    Product ID: 615b
    bcdDevice: 0104
    Bitstream serial number: 2a252adf30c460de

For additional device information use the --force-offline option.
```

But when I try to run the rubber ducky test script (copy/pasted from ReadTheDocs) I get a traceback error: 

```
$ python3 ./other-rubber-ducky.py 
INFO    | greatdancer    | Skipping GreatFET-based devices, as the greatfet python module isn't installed.
INFO    | moondancer     | Using the Moondancer backend.
INFO    | moondancer     | Connected FULL speed device 'facedancer.devices.keyboard.USBKeyboardDevice' to target host.
INFO    | other-rubber-ducky| Beginning message typing demo...
INFO    | moondancer     | Disconnecting from target host.
Traceback (most recent call last):
  File "/home/assess/Cynthion_Scripts/./other-rubber-ducky.py", line 25, in <module>
    main(device, type_letters())
  File "/home/assess/.local/lib/python3.10/site-packages/facedancer/devices/__init__.py", line 41, in default_main
    device.emulate(*coroutines)
  File "/home/assess/.local/lib/python3.10/site-packages/facedancer/device.py", line 228, in emulate
    self.run_with(*coroutines)
  File "/home/assess/.local/lib/python3.10/site-packages/facedancer/device.py", line 213, in run_with
    asyncio.run(inner())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/assess/.local/lib/python3.10/site-packages/facedancer/device.py", line 211, in inner
    await asyncio.gather(self.run(), *coroutines)
  File "/home/assess/.local/lib/python3.10/site-packages/facedancer/device.py", line 200, in run
    self.backend.service_irqs()
  File "/home/assess/.local/lib/python3.10/site-packages/facedancer/backends/moondancer.py", line 427, in service_irqs
    events: List[Tuple[int, int]] = self.api.get_interrupt_events()
  File "/home/assess/.local/lib/python3.10/site-packages/pygreat/comms.py", line 1126, in method
    return self.execute_command(verb_number, in_format, out_format, name=name, class_name=class_name,
  File "/home/assess/.local/lib/python3.10/site-packages/pygreat/comms.py", line 1281, in execute_command
    return self.comms_backend.execute_command(self.CLASS_NUMBER, verb, in_format,
  File "/home/assess/.local/lib/python3.10/site-packages/pygreat/comms.py", line 788, in execute_command
    raw_result = self.execute_raw_command(class_number, verb, payload, timeout,
  File "/home/assess/.local/lib/python3.10/site-packages/pygreat/comms_backends/usb1.py", line 403, in execute_raw_command
    self.device_handle.controlWrite(usb1.TYPE_VENDOR | usb1.RECIPIENT_DEVICE,
  File "/home/assess/.local/lib/python3.10/site-packages/usb1/__init__.py", line 1330, in controlWrite
    return self._controlTransfer(request_type, request, value, index, data,
  File "/home/assess/.local/lib/python3.10/site-packages/usb1/__init__.py", line 1307, in _controlTransfer
    mayRaiseUSBError(result)
  File "/home/assess/.local/lib/python3.10/site-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/home/assess/.local/lib/python3.10/site-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorTimeout: LIBUSB_ERROR_TIMEOUT [-7]
```

**Host Information**
Control Host: 

```
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
```

Target host: Apple M2 MacBook Pro running Sonoma 14.6.1 

I have also tried running it on a Kali 22.3 target host 

I am using cynthion+facedancer from the pip install.

I plugged the target host connection into the Controller host machine, ran `sudo dmesg -w`, and reran the script. 
Output:
`[ 1151.625350] usb 4-1: new full-speed USB device number 2 using xhci_hcd`

timokm, 2024-09-27:
Hello - I get the same errors using a "Ubuntu 22.04.3 LTS"  virtual machine (vmware)

using python v3.12.3 with cynthion+freedancer from pip install 

The host running the vm is windows 10 with vmware 17

[image]

[image]

[image]

KeuntaeShin, 2024-09-27:
@ All, I shared my case  in https://github.com/greatscottgadgets/cynthion/pull/179.

I had only one chance after reboot.

miek, 2024-09-27:
I can reproduce this too, and it seems like it might be an issue in the latest firmware/gateware - if I downgrade `cynthion` and re-run `cynthion run facedancer` the problem goes away.

KeuntaeShin, 2024-10-01:
> I can reproduce this too, and it seems like it might be an issue in the latest firmware/gateware - if I downgrade `cynthion` and re-run `cynthion run facedancer` the problem goes away.

Would you share the command to revert it? Plus, the version information.

KeuntaeShin, 2024-10-01:
> I've finally managed to reproduce this on my side but only when using Linux as the control host.
> 
> Will update as I learn more!

Would you share the OS you're using?

antoinevg, 2024-10-02:
I've pushed some fixes to https://github.com/greatscottgadgets/cynthion/pull/191 that should resolve this issue.

I'll try to do a new Cynthion release with these fixes next week but, before that, I'd greatly appreciate any feedback from anyone with the time to test it on their side!

timokm, 2024-10-02:
i would imagine that to test before the release i would have to build it from your commit right? 

if so then i will follow these instructions https://cynthion.readthedocs.io/en/latest/developer/soc_firmware_compilation.html   , let me know if i'm in the wrong path.

KeuntaeShin, 2024-10-03:
> I've pushed some fixes to #191 that should resolve this issue.
> 
> I'll try to do a new Cynthion release with these fixes next week but, before that, I'd greatly appreciate any feedback from anyone with the time to test it on their side!

I am willing to help the test. Please share the instruction.

closed 2024-10-08 by antoinevg