greatscottgadgets/cynthion #205: Cynthion standard Facedancer example with rubber ducky script hangs after INFO messages on Windows 11 Pro computer (build 26100.2605)

PMBNL, 2024-12-15:
I wanted my "hello world" experience with the default rubber ducky example on the website but I halts after these INFO messages:

(venv) C:\Users\Peter\OneDrive\GreatScottGadgets>python rubber-ducky.py
INFO    | __init__       | Starting emulation, press 'Control-C' to disconnect and exit.
INFO    | greatdancer    | Skipping GreatFET-based devices, as the greatfet python module isn't installed.
INFO    | moondancer     | Using the Moondancer backend.
INFO    | moondancer     | Connected FULL speed device 'facedancer.devices.keyboard.USBKeyboardDevice' to target host.

Tried both default examples to be sure because I saw minor differences:

https://cynthion.readthedocs.io/en/latest/getting_started_facedancer.html
https://facedancer.readthedocs.io/en/latest/getting_started.html

As you can see, Facedancer is loaded properly with the Cynthion device.

(venv) C:\Users\Peter\OneDrive\GreatScottGadgets>cynthion info
Cynthion version: 0.1.8
Apollo version: 1.1.1
Python version: 3.10.11 (tags/v3.10.11:7d4cc5a, Apr  5 2023, 00:38:17) [MSC v.1929 64 bit (AMD64)]

Found Apollo stub interface!
        Bitstream: Facedancer (Cynthion Project)
        Vendor ID: 1d50
        Product ID: 615b
        bcdDevice: 0104
        Bitstream serial number: xxxxxxx

antoinevg, 2024-12-17:
From your log output it looks like your Cynthion's `TARGET` port is not connecting to the target host. The full log output should have been something like this:

```
INFO    | __init__       | Starting emulation, press 'Control-C' to disconnect and exit.
INFO    | greatdancer    | Skipping GreatFET-based devices, as the greatfet python module isn't installed.
INFO    | moondancer     | Using the Moondancer backend.
INFO    | moondancer     | Connected FULL speed device 'facedancer.devices.keyboard.USBKeyboardDevice' to target host.
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | rubber-ducky   | Beginning message typing demo...
INFO    | moondancer     | Target host configuration complete.
```

Sometimes there can be issues with USB cables, do you have any other cables you could try for the Cynthion / Target Host connection?

Otherwise, can you please send me the debug log output from running:

```sh
python rubber-ducky.py -v 4
```

pbaard, 2024-12-17:
Thank you!!! It turned out to be a wiring issue. The scripts works now.

closed 2024-12-18 by antoinevg

stuartw1, 2025-01-26:
FYI I had a similar issue when using the Target A port to communicate with my test Android phone. None of my USB A to C cables worked, though the first USB C to C cable I tried worked fine.

Not sure what the problem was, though if anyone has Googled for the same thing that I have then it is one thing that you can try...