greatscottgadgets/cynthion #212: `cynthion setup --check`  should check other udev rules path as well

CoelacanthusHex, 2025-02-11:
In the current code, `cynthion setup --check` only check `/etc/udev/rules.d` but udev rules could be installed in other places especially when cynthion was installed as system package.
https://github.com/greatscottgadgets/cynthion/blob/f681b041b2b3d29a7b49f82bb2a20d67cd9b80a9/cynthion/python/src/commands/cynthion_setup.py#L14-L15

In `udev(7)`, it said:
> The udev rules are read from the files located in the system rules directories `/usr/lib/udev/rules.d` and `/usr/local/lib/udev/rules.d`, the volatile runtime directory `/run/udev/rules.d` and the local administration directory `/etc/udev/rules.d`. All rules files are collectively sorted and processed in lexical order, regardless of the directories in which they live. However, files with identical filenames replace each other. Files in /etc/ have the highest priority, files in /run/ take precedence over files with the same name under /usr/. This can be used to override a system-supplied rules file with a local file if needed; a symlink in /etc/ with the same name as a rules file in /usr/lib/, pointing to /dev/null, disables the rules file entirely. Rule files must have the extension .rules; other extensions are ignored.

So these places should be checked as well.

mossmann, 2025-02-20:
Have you encountered a specific problem caused by this issue?

CoelacanthusHex, 2025-02-20:
> Have you encountered a specific problem caused by this issue?

A false negative. I package cynthion as an Arch Linux packages, with Package Guideline, the udev rules were installed in /usr/lib/udev/rules.d. And then `cynthion setup --check` can't find udev rule and report "54-cynthion.rules not installed".

claui, 2025-04-18:
Author of the [`python-cynthion` PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=python-cynthion) here.

While @CoelacanthusHex may be right in that this is a minor cosmetic issue in `cynthion setup`, please keep in mind that fixing this issue would accomplish little:

- For regular users, fixing this would have no visible effect, because `/etc/udev/rules.d` is the only place that makes sense for them anyway.

- Users of the AUR package have nothing to gain by running `cynthion setup` or `cynthion setup --check`, because the udev rules have been taken care of already, and checking this is the only thing that `cynthion setup` currently does.

I’d say that this is, if anything, a packaging issue on the AUR package, and not a bug in the Cynthion project.

claui, 2025-05-20:
I have added a [patch](https://aur.archlinux.org/cgit/aur.git/tree/archlinux-managed-udev-rules.patch?h=python-cynthion&id=89e98e7a1cffb6c41ffc81b485819f0ecf31b540) to the latest [python-cynthion 0.2.0-1](https://aur.archlinux.org/packages/python-cynthion) PKGBUILD, which should fix the issue:

```plain
$ cynthion setup --check
Checking: Linux udev rules
✅	54-cynthion.rules is managed by Pacman
✅	54-cynthion.rules matches factory rules
✅	No unmanaged shadow copy of 54-cynthion.rules found

All checks completed successfully.
```

```plain
$ cynthion setup
Uninstalling: unmanaged Linux udev rules file
✅	/etc/udev/rules.d/54-cynthion.rules not present, skipping.

Uninstallation completed successfully.
```

Note that these changes are **not** meant to be upstreamed to the greatscottgadgets/cynthion repository.  
They only make sense for the (Pacman-managed) system package.

@CoelacanthusHex Let me know if the 0.2.0-1 version works for you.