greatscottgadgets/cynthion #220: usbproxy+filter moondancer.get_nak_status:failed to execute [EINVAL]

iBringsTheRuckus, 2025-03-11:
Cynthion version: 0.1.8
Apollo version: 1.1.1
Python version: 3.13.2 (main, Feb  5 2025, 01:23:35) [GCC 14.2.0]

Found Apollo stub interface!
        Bitstream: Facedancer (Cynthion Project)
        Vendor ID: 1d50
        Product ID: 615b
        bcdDevice: 0104

usbproxy.py
```
#!/usr/bin/env python3

from facedancer import main
from facedancer.proxy import USBProxyDevice
from facedancer.filters import USBProxySetupFilters, USBProxyPrettyPrintFilter

# Import attack filter
from bulk_payload_fuzzer import BulkPayloadFuzzer

# Set PNY flash drive vendor and product ID
ID_VENDOR = 0x154B
ID_PRODUCT = 0x1006

if __name__ == "__main__":
    proxy = USBProxyDevice(idVendor=ID_VENDOR, idProduct=ID_PRODUCT) 

    # Add base filters
    proxy.add_filter(USBProxySetupFilters(proxy, verbose=0))
    proxy.add_filter(USBProxyPrettyPrintFilter(verbose=5))

    # Enable Bulk Transfer Payload Fuzzing
    proxy.add_filter(BulkPayloadFuzzer())

    main(proxy)

bulk_payload_fuzzer.py
from facedancer.filters import USBProxyFilter

class BulkPayloadFuzzer(USBProxyFilter):
    def filter_in(self, endpoint, data):
        """Intercept and modify bulk read data (from flash drive to host)."""
        print(f"[DEBUG] Original Bulk Read Data: {data.hex()}")

        if len(data) > 0:
            # Corrupt the first 16 bytes of the bulk transfer
            corrupted_data = bytearray(data)
            for i in range(min(16, len(corrupted_data))):
                corrupted_data[i] ^= 0xFF  # Flip bits (bitwise XOR)
            
            print(f"[DEBUG] Modified Data: {corrupted_data.hex()}")
            return corrupted_data  # Return the corrupted data
        
        return data  # Otherwise, return unchanged data

    def filter_out(self, endpoint, data):
        """Intercept and modify bulk write data (from host to flash drive)."""
        print(f"[DEBUG] Original Bulk Write Data: {data.hex()}")

        if len(data) > 0:
            # Inject a fake boot sector identifier (possibly corrupting the FS)
            modified_data = data.replace(b'FAT32', b'XXXXXX')  # Corrupt FAT32 magic bytes
            print(f"[DEBUG] Modified Write Data: {modified_data.hex()}")
            return modified_data
        
        return data  # Otherwise, return unchanged data
```

Here is the output when I run usbproxy.py
[... 95 lines cut ...]
pygreat.comms.CommandFailureError: moondancer.get_nak_status:failed to execute [EINVAL]
```
Here is dmesg -W when running the script.
```
[ 1562.277672] usb-storage 4-4.4:1.0: USB Mass Storage device detected
[ 1562.277857] scsi host15: usb-storage 4-4.4:1.0
[ 1563.300040] scsi 15:0:0:0: Direct-Access     PNY      USB 3.2.1 FD     PMAP PQ: 0 ANSI: 6
[ 1563.300317] sd 15:0:0:0: Attached scsi generic sg6 type 0
[ 1563.301548] sd 15:0:0:0: [sdd] 242147328 512-byte logical blocks: (124 GB/115 GiB)
[ 1563.302345] sd 15:0:0:0: [sdd] Write Protect is off
[ 1563.302348] sd 15:0:0:0: [sdd] Mode Sense: 45 00 00 00
[ 1563.302956] sd 15:0:0:0: [sdd] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[ 1563.326165]  sdd: sdd1
[ 1563.326263] sd 15:0:0:0: [sdd] Attached SCSI removable disk
[ 3494.653884] usb-storage 4-4.4:1.0: USB Mass Storage device detected
[ 3494.654052] scsi host15: usb-storage 4-4.4:1.0
[ 3495.680316] scsi 15:0:0:0: Direct-Access     PNY      USB 3.2.1 FD     PMAP PQ: 0 ANSI: 6
[ 3495.680599] sd 15:0:0:0: Attached scsi generic sg6 type 0
[ 3495.681672] sd 15:0:0:0: [sdd] 242147328 512-byte logical blocks: (124 GB/115 GiB)
[ 3495.682462] sd 15:0:0:0: [sdd] Write Protect is off
[ 3495.682466] sd 15:0:0:0: [sdd] Mode Sense: 45 00 00 00
[ 3495.683059] sd 15:0:0:0: [sdd] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[ 3495.733868]  sdd: sdd1
[ 3495.733970] sd 15:0:0:0: [sdd] Attached SCSI removable disk
```

iBringsTheRuckus, 2025-03-19:
> From your logs:
> 
> WARNING | moondancer     | Moondancer only supports USB Full and High Speed. Ignoring requested speed: SUPER
> Cynthion only supports low, full and high speed USB. Can you try connecting the device being proxied to the computer through a USB-2.0 hub to force it to enumerate as a High speed device?

Thanks for looking into this. Would it be possible to use a 2.0 flashdrive on a 3.0 hub? I believe I have a couple old flash drives laying around instead of buying a new hub. If that's not possible, I'll buy a 2.0 hub and retest.

Also, what is the purpose of forcing a speed in code? Would that be an option as well?

Thanks

antoinevg, 2025-03-19:
A USB 2.0 flash drive will be fine even if it's on a 3.0 hub.

The ability to set the device speed in a Facedancer script is for the purposes of device emulation. USB Proxy itself will always try to match the speed the host enumerated the device as.

iBringsTheRuckus, 2025-03-19:
> A USB 2.0 flash drive will be fine even if it's on a 3.0 hub.
> 
> The ability to set the device speed in a Facedancer script is for the purposes of device emulation. USB Proxy itself will always try to match the speed the host enumerated the device as.

Understood and thanks for the explanation. I will reply with my findings as soon as I can.

iBringsTheRuckus, 2025-03-23:
I ran it again with a high speed flash drive here.
```
python usbproxy.py
INFO    | __init__       | Starting emulation, press 'Control-C' to disconnect and exit.
INFO    | greatdancer    | Skipping GreatFET-based devices, as the greatfet python module isn't installed.
INFO    | moondancer     | Using the Moondancer backend.
INFO    | moondancer     | Connected HIGH speed device 'facedancer.proxy.USBProxyDevice' to target host.
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | logging        | [21:39:30] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=256, index=0, length=64, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string=<facedancer.descriptor.StringRef object at 0x7f2d45ead7f0>, product_string=<facedancer.descriptor.StringRef object at 0x7f2d45e5f4d0>, serial_number_string=<facedancer.descriptor.StringRef object at 0x7f2d45e5f390>, supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=512, device_speed=None, requestable_descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7f2d4711f560>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7f2d44df3e00>))
INFO    | logging        | [21:39:30] <: b'\x12\x01\x00\x02\x00\x00\x00@\x81\x07\x06T\x00\x02\x01\x02\x03\x01'
INFO    | device         | Host issued a bus reset; resetting our connection.
INFO    | logging        | [21:39:30] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=256, index=0, length=18, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string=<facedancer.descriptor.StringRef object at 0x7f2d45ead7f0>, product_string=<facedancer.descriptor.StringRef object at 0x7f2d45e5f4d0>, serial_number_string=<facedancer.descriptor.StringRef object at 0x7f2d45e5f390>, supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=512, device_speed=None, requestable_descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7f2d4711f560>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7f2d44df3e00>))
INFO    | logging        | [21:39:30] <: b'\x12\x01\x00\x02\x00\x00\x00@\x81\x07\x06T\x00\x02\x01\x02\x03\x01'
INFO    | logging        | [21:39:30] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=512, index=0, length=9, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string=<facedancer.descriptor.StringRef object at 0x7f2d45ead7f0>, product_string=<facedancer.descriptor.StringRef object at 0x7f2d45e5f4d0>, serial_number_string=<facedancer.descriptor.StringRef object at 0x7f2d45e5f390>, supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=512, device_speed=None, requestable_descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7f2d4711f560>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7f2d44df3e00>))
INFO    | logging        | [21:39:30] <: b'\t\x02 \x00\x01\x01\x00\x80d'
INFO    | logging        | [21:39:30] USBControlRequest(direction=1, type=0, recipient=0, number=6, value=512, index=0, length=32, data=bytearray(b''), device=USBProxyDevice(name='generic device', device_class=0, device_subclass=0, protocol_revision_number=0, max_packet_size_ep0=64, vendor_id=24843, product_id=18003, manufacturer_string=<facedancer.descriptor.StringRef object at 0x7f2d45ead7f0>, product_string=<facedancer.descriptor.StringRef object at 0x7f2d45e5f4d0>, serial_number_string=<facedancer.descriptor.StringRef object at 0x7f2d45e5f390>, supported_languages=(<LanguageIDs.ENGLISH_US: 1033>,), device_revision=0, usb_spec_version=512, device_speed=None, requestable_descriptors={<DescriptorTypes.DEVICE: 1>: <function USBBaseDevice.__post_init__.<locals>.<lambda> at 0x7f2d4711f560>, <DescriptorTypes.CONFIGURATION: 2>: <bound method USBBaseDevice.get_configuration_descriptor of ...>, <DescriptorTypes.STRING: 3>: <bound method USBBaseDevice.get_string_descriptor of ...>}, configurations={}, backend=<facedancer.backends.moondancer.MoondancerApp object at 0x7f2d44df3e00>))
INFO    | logging        | [21:39:30] <: b'\t\x02 \x00\x01\x01\x00\x80d\t\x04\x00\x00\x02\x08\x06P\x00\x07\x05\x81\x02\x00\x02\x00\x07\x05\x02\x02\x00\x02\x00'
[... cut ...]
  File "/home/kali/cynthion_env/lib/python3.13/site-packages/facedancer/backends/moondancer.py", line 599, in handle_receive_packet                                                                             
    self.connected_device.handle_data_available(endpoint_number, data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/cynthion_env/lib/python3.13/site-packages/facedancer/proxy.py", line 165, in handle_data_available                                                                                           
    ep_num, data = f.filter_out(ep_num, data)
    ^^^^^^^^^^^^
ValueError: too many values to unpack (expected 2)
```