greatscottgadgets/cynthion #223: Cannot build Saturn-V

csBlueChip, 2025-03-24:
# Reproduce:
```
git clone https://github.com/greatscottgadgets/saturn-v
cd saturn-v
make
```
# Output:
```
...
/usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld: bootloader.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 108 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:94: bootloader.elf] Error 1
```
# Comment:
There is a comment in readthedocs that I may need to tweak options based on the version of my device ...but no (obvious) instructions on how to find this version number ...maybe that is my problem?

martinling, 2025-03-24:
We build Saturn-V using the [Arm GNU Toolchain](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads) packages.

This is currently the only way to get it to fit in 2KB, for some reason.

I've tried building it with several versions of the arm-none-eabi packages from Debian, and the binaries all come out a bit larger.

I'd love to know why, but I've not had time to investigate in detail.

csBlueChip, 2025-03-24:
Ahh. Yes, I am indeed building with standard Debian toolchains. Thank you. 

I'm also curious as to the root of that! ... That'll be project #457 lol

martinling, 2025-03-24:
I've opened https://github.com/greatscottgadgets/saturn-v/issues/23 to track this as it's come up several times.