greatscottgadgets/cynthion #224: HELP! Build Packetry Gateware & Build Moondancer SoC Gateware  Failed

zcn91, 2025-03-26:
Refer to https://github.com/greatscottgadgets/cynthion/BUILD.md for setting up the local development environment. In particular, ### Yosys Grab and install the latest toolchain from: https://github.com/YosysHQ/oss-cad-suite-build/releases/latest, I downloaded the corresponding version oss-cad-suite-linux-x64-20250324.tgz, but encountered errors when reaching Build Packetry Gateware & Build Moondancer Firmware, with a missing read_ilang prompt.
——————————————————————————————————
when I run this command
LUNA_PLATFORM="cynthion.gateware.platform:CynthionPlatformRev1D4" make analyzer
something is wrong , output as following:
python3 -m cynthion.gateware.analyzer.top --output build/analyzer.bit
INFO    | __init__    | Building for Cynthion r1.4...
ERROR: No such command: read_ilang (type 'help' for a command overview)
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/cynthion/gateware/analyzer/top.py", line 485, in <module>
    top_level_cli(USBAnalyzerApplet)
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/luna/__init__.py", line 122, in top_level_cli
    products = platform.build(fragment,
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/amaranth/build/plat.py", line 109, in build
    products = plan.execute_local(build_dir)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/amaranth/build/run.py", line 104, in execute_local
    subprocess.check_call(["sh", f"{self.script}.sh"],
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', 'build_top.sh']' returned non-zero exit status 1.
make: *** [Makefile:8: analyzer] Error 1
__________________________________________________________
when I run this command
LUNA_PLATFORM="cynthion.gateware.platform:CynthionPlatformRev1D4" make facedancer
has the same wrong , outpus as following:
mkdir -p build/
python3 -m cynthion.gateware.facedancer.top --output build/facedancer.bit
/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/cynthion/gateware/facedancer/top.py:13: DeprecationWarning: instead of `amaranth.hdl.rec.Record`, use the `amaranth.lib.data` and `amaranth.lib.wiring` libraries as appropriate for the application; `amaranth.hdl.rec` will be removed in Amaranth 0.6
  from amaranth.hdl.rec         import Record
INFO    | top         | Building for <cynthion.gateware.platform.cynthion_r1_4.CynthionPlatformRev1D4 object at 0x7b821b44c550> with clock frequency: 60000000
INFO    | vexriscv    | reading verilog file: /home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/luna_soc/gateware/cpu/verilog/vexriscv/vexriscv_cynthion+jtag.v
INFO    | top_level_cli| Building for Cynthion r1.4...
INFO    | top_level_cli| Build directory: /tmp/tmp582_foaz
DEBUG   | core        | Skipping resource led/6, as it is not present on this platform.
DEBUG   | core        | Skipping resource led/7, as it is not present on this platform.
ERROR: No such command: read_ilang (type 'help' for a command overview)
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/cynthion/gateware/facedancer/top.py", line 328, in <module>
    top_level_cli(_design, **_overrides)
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/luna_soc/top_level_cli.py", line 152, in top_level_cli
    build(args, fragment, platform, build_dir)
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/luna_soc/top_level_cli.py", line 181, in build
    products = _build_gateware(args, fragment, platform, build_dir)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/luna_soc/top_level_cli.py", line 210, in _build_gateware
    products = platform.build(
               ^^^^^^^^^^^^^^^
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/amaranth/build/plat.py", line 109, in build
    products = plan.execute_local(build_dir)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/amaranth/build/run.py", line 104, in execute_local
    subprocess.check_call(["sh", f"{self.script}.sh"],
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', 'build_top.sh']' returned non-zero exit status 1.
make: *** [Makefile:16: facedancer] Error 1

 Has anyone experienced this issue, and how should it be resolved? Thanks! I have changed the device "LFE5U-12F" to "LFE5U-25F" in cynthion_r1_4.py Refer to https://github.com/greatscottgadgets/cynthion/issues/185

antoinevg, 2025-03-26:
The most recent Yosys release that works with Amaranth 0.4.x is:

https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2024-11-01

We're currently in the last stages of merging the work needed to bring us up to date with the latest Amaranth 0.5.x releases.

Related:

* https://github.com/greatscottgadgets/cynthion/issues/214
* https://github.com/greatscottgadgets/cynthion/issues/213
* https://github.com/greatscottgadgets/luna/issues/284
* https://github.com/greatscottgadgets/luna/pull/278
* https://github.com/greatscottgadgets/luna-soc/pull/33

zcn91, 2025-03-27:
thanks , but  when I run this command
LUNA_PLATFORM="cynthion.gateware.platform:CynthionPlatformRev1D4" make analyzer
something is wrong another , output as following:
⦗OSS CAD Suite⦘ (gsg-cynthion) zzz@zzz-virtual-machine:~/Desktop/cynthion.git/cynthion/python$     LUNA_PLATFORM="cynthion.gateware.platform:CynthionPlatformRev1D4" make analyzer
python3 -m cynthion.gateware.analyzer.top --output build/analyzer.bit
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/cynthion/gateware/analyzer/top.py", line 19, in <module>
    from amaranth                            import Signal, Elaboratable, Module, DomainRenamer, ResetInserter
ImportError: cannot import name 'Signal' from 'amaranth' (unknown location)
make: *** [Makefile:8: analyzer] Error 1
⦗OSS CAD Suite⦘ (gsg-cynthion) zzz@zzz-virtual-machine:~/Desktop/cynthion.git/cynthion/python$ pip show amaranth
Name: amaranth
Version: 0.4.1
Summary: Amaranth hardware definition language
Home-page: 
Author: Amaranth HDL contributors
Author-email: 
License: Copyright (C) 2019-2023 Amaranth HDL contributors

Has anyone experienced this issue, and how should it be resolved? Thanks! I have changed the device "LFE5U-12F" to "LFE5U-25F" in cynthion_r1_4.py ,and now amaranth=0.4.1

zcn91, 2025-03-27:
**I tried two different comands and found the following results.**
1),first one,hasv error:
LUNA_PLATFORM="cynthion.gateware.platform:CynthionPlatformRev1D4"
 python3 -m cynthion.gateware.facedancer.top
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/zzz/Documents/Yosys/oss-cad-suite/lib/python3.11/site-packages/cynthion/gateware/facedancer/top.py", line 11, in <module>
    from amaranth                 import Cat, DomainRenamer, Elaboratable, Module, ResetSignal, Signal, Record
ImportError: cannot import name 'Cat' from 'amaranth' (unknown location)
2),the second one ,no error:
   LUNA_PLATFORM="cynthion.gateware.platform:CynthionPlatformRev1D4" make analyzer
mkdir -p build/
python -m cynthion.gateware.analyzer.top --output build/analyzer.bit
INFO    | __init__    | Building for Cynthion r1.4...
INFO    | __init__    | Build complete.
3)
(gsg-cynthion) ⦗OSS CAD Suite⦘ zzz@zzz-virtual-machine:~/Desktop/cynthion.git/cynthion/python$ python
Python 3.11.11 (main, Mar 24 2025, 15:20:25) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> exit()
(gsg-cynthion) ⦗OSS CAD Suite⦘ zzz@zzz-virtual-machine:~/Desktop/cynthion.git/cynthion/python$ python3
Python 3.11.6 (tags/v3.11.6:8b6ee5ba3b, Jan 17 2024, 08:09:36) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> exit()
(gsg-cynthion) ⦗OSS CAD Suite⦘ zzz@zzz-virtual-machine:~/Desktop/cynthion.git/cynthion/python$ which python
/home/zzz/.pyenv/shims/python
(gsg-cynthion) ⦗OSS CAD Suite⦘ zzz@zzz-virtual-machine:~/Desktop/cynthion.git/cynthion/python$ which python3
(gsg-cynthion) ⦗OSS CAD Suite⦘ zzz@zzz-virtual-machine:~/Desktop/cynthion.git/cynthion/python$ 

(gsg-cynthion) ⦗OSS CAD Suite⦘ zzz@zzz-virtual-machine:~/Desktop/cynthion.git/cynthion/python$ 

(gsg-cynthion) ⦗OSS CAD Suite⦘ zzz@zzz-virtual-machine:~/Desktop/cynthion.git/cynthion/python$ 
(gsg-cynthion) ⦗OSS CAD Suite⦘ zzz@zzz-virtual-machine:~/Desktop/cynthion.git/cynthion/python$ which python3
/home/zzz/Documents/Yosys/oss-cad-suite/py3bin/python3
____________________________________________________________________________
**officalially provided makefile are located at cynthion.git/cynthion/python/makefile, it's content as follows:**
SHELL := /bin/zsh

UART	?= /dev/ttyACM0

# - gateware ------------------------------------------------------------------

analyzer:
	python3 -m cynthion.gateware.analyzer.top --output build/$@.bit

selftest:
	mkdir -p build/
	python3 -m cynthion.gateware.selftest.top --output build/$@.bit

facedancer:
	mkdir -p build/
	python3 -m cynthion.gateware.facedancer.top --output build/$@.bit

svd:
	mkdir -p build/gensvd/
	python3 -m cynthion.gateware.facedancer.top --generate-svd > build/gensvd/lunasoc.svd
**But if I modify it to the following  analyzer.bit can build success:**
SHELL := /usr/local/bin/zsh

UART	?= /dev/ttyACM0

# - gateware ------------------------------------------------------------------

analyzer:
	mkdir -p build/
	python -m cynthion.gateware.analyzer.top --output build/$@.bit

selftest:
	mkdir -p build/
	python3 -m cynthion.gateware.selftest.top --output build/$@.bit

facedancer:
	mkdir -p build/
	python3 -m cynthion.gateware.facedancer.top --output build/$@.bit

svd:
	mkdir -p build/gensvd/
	python3 -m cynthion.gateware.facedancer.top --generate-svd > build/gensvd/lunasoc.svd
____________________________________________________________________________________________________________

**my question is , when compiling other bit firmware , should it be modified like this?**

zcn91, 2025-03-28:
Thank you, it was very helpful for me, I have generated anaylzer.bit and facedancer.bit, and I am ready to flash the bitstream to my board

zcn91, 2025-03-31:
**I  changed the device "LFE5U-12F" to "LFE5U-25F" in cynthion_r1_4.py,because my device is "LFE5U-25F" .
class CynthionPlatformRev1D4(CynthionPlatform):**
    """ Board description for Cynthion r1.4 """

    name        = "Cynthion r1.4"
    version     = (1, 4)
    device      = "LFE5U-52F"
    package     = "BG256"
    speed       = os.getenv("ECP5_SPEED_GRADE", "8")
**base on this(cynthion_r1_4.py), I have generate bitstream as follows:**
⦗OSS CAD Suite⦘ (gsg-cynthion) zzz@zzz-virtual-machine:~/Desktop/cynthion.git/cynthion/python$  LUNA_PLATFORM="cynthion.gateware.platform:CynthionPlatformRev1D4" make analyzer
mkdir -p build/
python -m cynthion.gateware.analyzer.top --output build/analyzer.bit
INFO    | __init__    | Building for Cynthion r1.4...
INFO    | __init__    | Build complete.

**so, I am ready to flash this bitstream to my board,but has some errors,how to solve it?**
(gsg-cynthion) zzz@zzz-virtual-machine:~/Desktop/cynthion.git/cynthion/python/build$ cynthion flash --bitstream analyzer.bit 
Traceback (most recent call last):
  File "/home/zzz/.pyenv/versions/gsg-cynthion/bin/cynthion", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/zzz/.pyenv/versions/3.11.11/envs/gsg-cynthion/lib/python3.11/site-packages/cynthion/commands/cli.py", line 94, in main
    device = ApolloDebugger(force_offline=force_offline)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zzz/.pyenv/versions/3.11.11/envs/gsg-cynthion/lib/python3.11/site-packages/apollo_fpga/__init__.py", line 105, in __init__
    device = self._find_device(self.APOLLO_USB_IDS, custom_match=self._device_has_apollo_id)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zzz/.pyenv/versions/3.11.11/envs/gsg-cynthion/lib/python3.11/site-packages/apollo_fpga/__init__.py", line 190, in _find_device
    device = usb.core.find(backend=cls.backend, idVendor=vid, idProduct=pid, custom_match=custom_match)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zzz/.pyenv/versions/3.11.11/envs/gsg-cynthion/lib/python3.11/site-packages/usb/core.py", line 1327, in find
    return next(device_iter(**args))
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zzz/.pyenv/versions/3.11.11/envs/gsg-cynthion/lib/python3.11/site-packages/usb/core.py", line 1307, in device_iter
    if all(tests) and (custom_match is None or custom_match(d)):
                                               ^^^^^^^^^^^^^^^
  File "/home/zzz/.pyenv/versions/3.11.11/envs/gsg-cynthion/lib/python3.11/site-packages/apollo_fpga/__init__.py", line 239, in _device_has_apollo_id
    response = device.ctrl_transfer(request_type, ApolloDebugger.REQUEST_GET_ID, data_or_wLength=256, timeout=500)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zzz/.pyenv/versions/3.11.11/envs/gsg-cynthion/lib/python3.11/site-packages/usb/core.py", line 1093, in ctrl_transfer
    ret = self._ctx.backend.ctrl_transfer(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zzz/.pyenv/versions/3.11.11/envs/gsg-cynthion/lib/python3.11/site-packages/usb/backend/libusb1.py", line 897, in ctrl_transfer
    ret = _check(self.lib.libusb_control_transfer(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zzz/.pyenv/versions/3.11.11/envs/gsg-cynthion/lib/python3.11/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out

antoinevg, 2025-04-22:
It's possible that the FPGA is not handing off control from the Analyzer bitstream to Apollo.

Can you try pressing Cynthion's PROGRAM button once before you try `cynthion run analyzer` ?