greatscottgadgets/cynthion #226: The same device both visible and not visible

lab313ru, 2025-03-31:
My device doesn't show up while connected via Cynthion in one of two modes.

Visible mode:
```
Bus 007 Device 031: ID 2207:330d Fuzhou Rockchip Electronics Company USB download gadget
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2207 Fuzhou Rockchip Electronics Company
  idProduct          0x330d USB download gadget
  bcdDevice           99.99
  iManufacturer           1 Rockchip
  iProduct                2 USB download gadget
  iSerial                 3 01119F632DED4912
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          2 USB download gadget
    bmAttributes         0xc0
      Self Powered
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      6 [unknown]
      bInterfaceProtocol      5 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0008
  bNumDeviceCaps          1
  ** UNRECOGNIZED:  03 10 00
Device Status:     0x0001
  Self Powered
```

Not visible mode:
```
Bus 007 Device 030: ID 2207:330e Fuzhou Rockchip Electronics Company 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2207 Fuzhou Rockchip Electronics Company
  idProduct          0x330e 
  bcdDevice            1.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              400mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      6 [unknown]
      bInterfaceProtocol      5 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
cannot read device status, Resource temporarily unavailable (11)
```

Cable connections are the same. Direct connection of my device to the PC works as usual in two modes. What can be wrong? How can I help in resolving that?

shidorun, 2025-04-01:
У вашего второго девайса скорее всего завышена паразитная ёмкость между выводами UD+ и UD-. Возможно это результат плохой пайки разъёмов или платы девайса. При подключении напрямую Виндовс определяет устройство правильно и паразитная ёмкость хоть и большая но в пределах допустимого, а при внесении дополнительной паразитной ёмкости в виде снифера и длинных кабелей происходит завал фронтов и возникают ошибки CRC в пакетах. Виндовс откидывает повреждённые пакеты, но видит что устройство есть на шине и пишет что девайс не опознался правильно.
Попробуйте максимально укоротить USB кабели. Сделайте самостоятельно небольшие отрезки по 20-30 см. Я изготавливал себе такие из старых оригинальных от HP-принтера. Кабель должен быть как можно толще. Тогда у него внутри реально витая пара в хорошем экране а не просто пучок проводов в пластике.
Возможно сам снифер вносит слишком много ёмкости. 
У меня Teledyne LeCroy и на коротких кабелях никогда проблем нет. На длинных 2-3 метра были.

lab313ru, 2025-04-01:
But the device is the same with the same cableing. But when it exposes 0x330e PID it isn't visible via Cynthion, and when it exposes 0x330D - it is visible.

shidorun, 2025-04-01:
There are 2 modes inside the device. In the second mode, other circuits on the device board are probably involved. Therefore, parasitic capacitance is added. This addition is critical in your case. To check this version, just shorten the USB cables as much as possible.

martinling, 2025-04-01:
Unless the device draws a lot of current, the issue is unlikely to be with passing through power. The VBUS resistance between Cynthion ports is very low, approximately 0.01Ω, and this is verified in our factory tests of each unit.

The problem is more likely with signal integrity on the data lines. Even though one mode works, it may be the case that the cumulative effects of the cables, the Cynthion passthrough, and perhaps also poor receiver performance on the target device and/or host, are collectively just enough to make signal integrity marginal in this setup. Then some otherwise irrelevant difference between the two modes, such as a specific data pattern, may result in a failure, even though all the individual parts of the signal chain work OK elsewhere.

Unless you are already very confident of the internal structure of the cables you are using, I would suggest looking for some cables that are known to be of the proper internal construction for good signal integrity. At a minimum that means full shielding, properly connected to the connector shells at both ends. There are surprisingly many cables on the market that do not meet these basic requirements including some from popular brands, with very attractive braided fabric sleeves and fancy connectors, but poor signal integrity.

lab313ru, 2025-04-02:
In result I've cut one cable by half and now it works:)

martinling, 2025-04-02:
Great! Glad you were able to solve it.

closed 2025-04-02 by martinling

shidorun, 2025-04-02:
The best advice is to make custom cables from old HP printer cables from the 1990s. They have always been of reliable quality, and there are still plenty available. Create short segments of about 20-30 cm. Your issue is a common one, everyone goes through this. 
Thank you for following my advice.
Vlad, I need your help. How can I contact you?

lab313ru, 2025-04-02:
I've specified email in my acc bio, check it out.