greatscottgadgets/cynthion #229: Still the read_ilang problem

lab313ru, 2025-04-05:
What I've done:
1. Downloaded the latest oss-cad-suite release
2. git cloned latest cynthion
3. `source ~/oss-cad-suite/environment`
4. `pip3 uninstall -y -r <(pip3 freeze)`
5. `cd ~/cynthion/cynthion/python/`
6. `pip3 install --upgrade ".[gateware]"`
7. `LUNA_PLATFORM="cynthion.gateware.platform:CynthionPlatformRev1D4" make analyzer`

In result I'm getting the same as in previous issues:
```
python3 -m cynthion.gateware.analyzer.top --output build/analyzer.bit
INFO    | __init__    | Building for Cynthion r1.4...
ERROR: No such command: read_ilang (type 'help' for a command overview)
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/vk/oss-cad-suite/lib/python3.11/site-packages/cynthion/gateware/analyzer/top.py", line 485, in <module>
    top_level_cli(USBAnalyzerApplet)
  File "/home/vk/oss-cad-suite/lib/python3.11/site-packages/luna/__init__.py", line 122, in top_level_cli
    products = platform.build(fragment,
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vk/oss-cad-suite/lib/python3.11/site-packages/amaranth/build/plat.py", line 109, in build
    products = plan.execute_local(build_dir)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vk/oss-cad-suite/lib/python3.11/site-packages/amaranth/build/run.py", line 104, in execute_local
    subprocess.check_call(["sh", f"{self.script}.sh"],
  File "/home/vk/oss-cad-suite/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', 'build_top.sh']' returned non-zero exit status 1.
make: *** [Makefile:8: analyzer] Error 1

```

What I'm doing wrong?

miek, 2025-04-05:
Updating everything to use Amaranth 0.5 and fix that issue is still in progress. We've merged the first steps in https://github.com/greatscottgadgets/luna/pull/278 and https://github.com/greatscottgadgets/luna-soc/pull/33, but there's still https://github.com/greatscottgadgets/cynthion/pull/193 which is in review at the moment. You could try out that branch, or use an older oss-cad-suite in the meantime.

lab313ru, 2025-04-05:
Which `oss-cad-suite` is the last compatible with cythion?

miek, 2025-04-05:
> Which `oss-cad-suite` is the last compatible with cythion?

https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2024-11-01

lab313ru, 2025-04-05:
Thanks, now I can compile and use Blinky:)

closed 2025-04-17 by martinling