greatscottgadgets/cynthion #231: No documentation for power cycling the targets

HunterDOmar, 2025-04-08:
I want to use Cynthion to power cycle the device using python api but I can't find any documentation or api in the repo to do this .
help is much appreciated

tieknimmers, 2025-04-08:
I was actually looking into the same thing today. I've connected the target at `TARGET A`, host at `TARGET C` and the host also at `CONTROL`. I made a gateware that power cycles the target based on the state of pin 1 of PMOD A.

```python
#!/usr/bin/env python3
from amaranth import *

class Top(Elaboratable):
    def elaborate(self, platform):
        m = Module()

        # request required resources
        usb_enable = platform.request("target_c_vbus_en")
        pmod0 = platform.request("user_pmod", 0)
        
        # check pin 1 of pmod A to enable/disable usb connection
        with m.If(pmod0.i[0] == 1):
            m.d.sync += usb_enable.o.eq(1)
        with m.Else():
            m.d.sync += usb_enable.o.eq(0)
        return m

if __name__ == "__main__":
    from luna import top_level_cli
    top_level_cli(Top)
```
Note, I'm not 100% sure if this is the intended or most efficient approach is, but it seems to work. When pin 1 of PMOD A is connected to ground the target is off and when it's connected to 3v3 it's on. Both the ground and 3v3 signal are available on PMOD A. Any feedback on this approach is welcome.

martinling, 2025-04-09:
That's a good example of how this can be done from gateware. Note that it can be simplified a little by replacing:

```python
with m.If(pmod0.i[0] == 1):
    m.d.sync += usb_enable.o.eq(1)
with m.Else():
    m.d.sync += usb_enable.o.eq(0)
```
with
```python
m.d.sync += usb_enable.o.eq(pmod0.i[0])
```