greatscottgadgets/cynthion #235: `cynthion update` error on Windows 11 / Python 3.13.3 / micromamba

LispEngineer, 2025-04-26:
Hi. I just received a brand new Cynthion and went to update the firmware. The updater seems to have failed - or succeeded. I don't know!

I'm using micromamba with Python 3.13.3 on Windows 11. I installed cynthion per instructions, then tested it:

`cynthion info --force-offline`
```
Cynthion version: 0.1.8
Apollo version: 1.1.1
Python version: 3.13.3 | packaged by conda-forge | (main, Apr 14 2025, 20:31:24) [MSC v.1943 64 bit (AMD64)]

Found Apollo stub interface!
        Bitstream: USB Analyzer (Cynthion Project)
        Vendor ID: 1d50
        Product ID: 615b
        bcdDevice: 0104
        Bitstream serial number: 277371df30c460de

Forcing offline.

Found Cynthion device!
        Hardware: Cynthion r1.4
        Manufacturer: Great Scott Gadgets
        Product: Apollo Debugger
        Serial number: LYQZWLLMGJJVCIBAEAMANEQMQN
        Vendor ID: 1d50
        Product ID: 615c
        bcdDevice: 0104
        Firmware version: v1.0.2
        USB API version: 1.0
        Flash UID: ffffffffffffffff
```

Then I ran the update and got the error below:

`cynthion update`

```
Updating FPGA configuration flash with 251871 bytes...
Operation complete!
Updating device firmware with 13888 bytes...
Operation complete!
Traceback (most recent call last):
  File "C:\Users\Doug\micromamba\envs\cynthion_env\Lib\weakref.py", line 666, in _exitfunc
    f()
    ~^^
  File "C:\Users\Doug\micromamba\envs\cynthion_env\Lib\weakref.py", line 590, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
           ~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Doug\micromamba\envs\cynthion_env\Lib\site-packages\usb\_objfinalizer.py", line 102, in _do_finalize_object_ref
    obj._do_finalize_object()
    ~~~~~~~~~~~~~~~~~~~~~~~^^
  File "C:\Users\Doug\micromamba\envs\cynthion_env\Lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
    ~~~~~~~~~~~~~~~~~~~~~^^
  File "C:\Users\Doug\micromamba\envs\cynthion_env\Lib\site-packages\usb\backend\libusb1.py", line 616, in _finalize_object
    _lib.libusb_unref_device(self.devid)
    ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
OSError: exception: access violation writing 0x0000000000000000
```

After the update command, Windows did it's usual "bleep" indicating that something disconnected. Presumably it was the Cynthion rebooting with the new firmware?

I re-ran the info command above afterwards (did not unplug/replug the Cynthion):

```
Cynthion version: 0.1.8
Apollo version: 1.1.1
Python version: 3.13.3 | packaged by conda-forge | (main, Apr 14 2025, 20:31:24) [MSC v.1943 64 bit (AMD64)]

Found Apollo stub interface!
        Bitstream: USB Analyzer (Cynthion Project)
        Vendor ID: 1d50
        Product ID: 615b
        bcdDevice: 0104
        Bitstream serial number: 277371df30c460de

Forcing offline.

Found Cynthion device!
        Hardware: Cynthion r1.4
        Manufacturer: Great Scott Gadgets
        Product: Cynthion Apollo Debugger
        Serial number: LY7ZWL2MGJJVCIBAEA3A4EQM74
        Vendor ID: 1d50
        Product ID: 615c
        bcdDevice: 0104
        ADC reading: 3212
        Firmware version: v1.1.1
        USB API version: 1.2
        Flash UID: 277371df30c460de
```

I would assume this error is not supposed to happen. I'm not sure if both the updates ran correctly. Thanks.

martinling, 2025-05-02:
The update succeeded, but you ran into a crash while it was shutting down afterwards.

The issue causing the crash is being tracked at https://github.com/greatscottgadgets/pyfwup/issues/14.

LispEngineer, 2025-05-02:
Thank you. Closing this issue as a duplicate then.

closed 2025-05-02 by LispEngineer