greatscottgadgets/cynthion #236: OSError: Configuration failed: part ID mismatch / (04a00e10)

robocop4, 2025-05-11:
Hello. When I connect the device via USB, `cynthion` install and `cynthion setup`, I run the `cynthion info` command as shown in the following snippet from the terminal:

```bash
cynthion info
Cynthion version: 0.1.8
Apollo version: 1.1.1
Python version: 3.9.1 (default, May 11 2025, 11:48:30) 
[GCC 12.2.0]

Found Cynthion device!
	Hardware: Cynthion r1.4
	Manufacturer: Apollo Project
	Product: Apollo Debugger
	Serial number: RTTKXFKZJNKFCIBAEAYVWJQN74
	Vendor ID: 1d50
	Product ID: 615c
	bcdDevice: 0104
	ADC reading: 883
	Firmware version: v1.1.1
	USB API version: 1.2
```

Further I try to launch the analyzer and encounter an error as shown in the following code fragment:

```bash
cynthion run analyzer
Uploading target bitstream to FPGA with 251871 bytes...
Traceback (most recent call last):
  File "/home/admin0/Desktop/Cynthion/env/bin/cynthion", line 8, in <module>
    sys.exit(main())
  File "/home/admin0/Desktop/Cynthion/env/lib/python3.9/site-packages/cynthion/commands/cli.py", line 95, in main
    args.func(device, args)
  File "/home/admin0/Desktop/Cynthion/env/lib/python3.9/site-packages/cynthion/commands/cynthion_run.py", line 21, in cynthion_run
    run_bitstream(device, find_cynthion_bitstream(device, f"analyzer.bit"))
  File "/home/admin0/Desktop/Cynthion/env/lib/python3.9/site-packages/cynthion/commands/util.py", line 197, in run_bitstream
    programmer.configure(bitstream)
  File "/home/admin0/Desktop/Cynthion/env/lib/python3.9/site-packages/apollo_fpga/ecp5.py", line 464, in configure
    self._validate_status(status, expect_done=True)
  File "/home/admin0/Desktop/Cynthion/env/lib/python3.9/site-packages/apollo_fpga/ecp5.py", line 364, in _validate_status
    error_handler("Configuration failed: {} / ({:08x})".format(self.STATUS_ERROR_CODES[error_code], status))
  File "/home/admin0/Desktop/Cynthion/env/lib/python3.9/site-packages/apollo_fpga/ecp5.py", line 318, in raise_error
    raise IOError(msg)
OSError: Configuration failed: part ID mismatch / (04a00e10)
```

If I run `cynthion flash analyzer` it succeeds but the device does not go into usb analyzer mode and `cynthion info` still returns `Product: Apollo Debugger`.

martinling, 2025-05-15:
It appears that your Cynthion board wasn't made by Great Scott Gadgets, and whoever made it has populated it with an LFE5U-25F FPGA, rather than the LFE5U-12F part that is specified in our BOM and in the Cynthion r1.4 platform file. When you run `cynthion run analyzer`, it tries to load our pre-built bitstream to the FPGA, and fails because the bitstream has the wrong part ID for the chip.

You should be able to build an analyzer bitstream suitable for your board by modifying the cynthion Python source, changing `LFE5U-12F` to `LFE5U-25F` in cynthion_r1.4.py, re-installing the package and then running `python -m cynthion.gateware.analyzer.top -o analyzer.bit`

If you're having further trouble, please contact the seller you bought the board from for support. We're happy to provide support for our open source software and designs, but we can't provide support for hardware that we didn't manufacture.

We recommend folks buy from our authorized distributors, who are listed at https://greatscottgadgets.com/cynthion/. This helps support our work. Thanks!

robocop4, 2025-05-17:
Thank you for your detailed response. I spent a few days and built a docker container that allows to build firmware for cynthion. The following code snippet shows the result of my work:

```
FROM ubuntu:20.04 AS ubuntu
ENV DEBIAN_FRONTEND=noninteractive
ENV LANG=en_US.UTF-8
ENV LANGUAGE=en_US:en
ENV LC_ALL=en_US.UTF-8

RUN apt-get update && apt-get install -y git wget tar curl  python3.9 python3.9-venv
# Adding backend. Here you can change the revision 
ENV LUNA_PLATFORM="cynthion.gateware.platform:CynthionPlatformRev1D4"

RUN mkdir -p /workspace/oss-cad-suite && chmod -R 755 /workspace

WORKDIR /workspace

RUN git clone https://github.com/greatscottgadgets/cynthion.git

#According to the following issue this is the latest version that works at the moment:
#https://github.com/greatscottgadgets/cynthion/issues/224
RUN curl -L https://github.com/YosysHQ/oss-cad-suite-build/releases/download/2024-11-01/oss-cad-suite-linux-x64-20241101.tgz \
    | tar -xz -C /workspace/oss-cad-suite --strip-components=1

ENV PATH="/workspace/oss-cad-suite/bin:${PATH}"

RUN python3.9 -m venv /workspace/env
RUN /workspace/env/bin/pip install --upgrade pip setuptools setuptools-scm
WORKDIR /workspace/cynthion/cynthion/python/ 
#https://github.com/greatscottgadgets/cynthion/issues/185
RUN sed -i 's/device *= *"LFE5U-12F"/device      = "LFE5U-25F"/' /workspace/cynthion/cynthion/python/src/gateware/platform/cynthion_r1_4.py
RUN /workspace/env/bin/pip install --editable .  

RUN echo '#!/bin/bash' > /workspace/cynthion/cynthion/python/runbuild.sh && \
echo 'if [[ "$1" == "a" ]]; then' >> /workspace/cynthion/cynthion/python/runbuild.sh && \
echo '    /workspace/env/bin/python3.9 -m cynthion.gateware.analyzer.top -D --output /output/analyzer_1d4_25F.bit' >> /workspace/cynthion/cynthion/python/runbuild.sh && \
echo 'else' >> /workspace/cynthion/cynthion/python/runbuild.sh && \
echo '    /workspace/env/bin/python3.9 -m cynthion.gateware.facedancer.top -D --output /output/facedancer_1d4_25F.bit' >> /workspace/cynthion/cynthion/python/runbuild.sh && \
echo 'fi' >> /workspace/cynthion/cynthion/python/runbuild.sh && \
chmod +x /workspace/cynthion/cynthion/python/runbuild.sh

#docker run -it -v "$(pwd)":/output cynthion /bin/bash
```

To build the image, the following command can be used:
`docker build --no-cache -t cynthion .`

Once built, the container must be started using the following command:

`docker run -it -v "$(pwd)":/output cynthion /bin/bash`

This command will start the Docker container and create a shared folder, allowing you to transfer the final build to your host.

Inside the Docker container, run the `runbuild.sh` script with argemounts `a` or `f` depending on whether you want to build analyzer or facedancer firmware. After the job is finished on your host, the corresponding file will appear in the folder from where you started the container. 

Use the `cynthion flash --bitstream analyzer_1d4_25F.bit` command to flash your device. 
This assembly method is designed for the `LFE5U-25F` device. If you want to build for `LFE5U-12F`, delete the following line in the Dockerfile:

`RUN sed -i 's/device *= *"LFE5U-12F"/device      = "LFE5U-25F"/' /workspace/cynthion/cynthion/python/src/gateware/platform/cynthion_r1_4.py`

Hope this helps someone save some time.