greatscottgadgets/cynthion #237: update error: Device not found after DFU_DETACH

gniezen, 2025-06-19:
Hi there,

When updating the firmware using `cynthion update --mcu` I'm getting

```
cynthion update --mcu
Traceback (most recent call last):
  File "/home/gerrit/.local/bin/cynthion", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/gerrit/.local/share/pipx/venvs/cynthion/lib/python3.12/site-packages/cynthion/commands/cli.py", line 95, in main
    args.func(device, args)
  File "/home/gerrit/.local/share/pipx/venvs/cynthion/lib/python3.12/site-packages/cynthion/commands/cynthion_update.py", line 18, in cynthion_update
    flash_mcu_firmware(device, find_cynthion_asset("apollo.bin"))
  File "/home/gerrit/.local/share/pipx/venvs/cynthion/lib/python3.12/site-packages/cynthion/commands/util.py", line 157, in flash_mcu_firmware
    board = DFUTarget(idVendor=shared.usb.bVendorId.apollo, idProduct=shared.usb.bProductId.apollo)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gerrit/.local/share/pipx/venvs/cynthion/lib/python3.12/site-packages/fwup/dfu.py", line 194, in __init__
    raise BoardNotFoundError("Device not found after DFU_DETACH.")
fwup.errors.BoardNotFoundError: Device not found after DFU_DETACH.

```

I'm using Ubuntu 24.04.2 LTS.

gniezen, 2025-07-17:
Yes, that works! I first checked that it still failed with the same error after I managed to update the firmware on a Mac, added the timeout as suggested and

```
cynthion update --mcu
Updating device firmware with 13888 bytes...
Operation complete!
```

mndza, 2025-09-10:
The fix is included in pyfwup 0.5.3, so I’m closing this issue. Please reopen it if you encounter the problem again.