greatscottgadgets/cynthion #244: cynthion update error: "usb.core.USBError: [Errno 2] Entity not found"

tzuyin, 2025-07-21:
Hi there,

When updating using "cynthion update" in via PowerShell with Administrator privilege on Windows 11 64-bit, I'm getting

> PS C:\Windows\System32> cynthion update
> Updating FPGA configuration flash with 242788 bytes...
> Traceback (most recent call last):
>   File "<frozen runpy>", line 198, in _run_module_as_main
>   File "<frozen runpy>", line 88, in _run_code
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\Scripts\cynthion.exe\__main__.py", line 7, in <module>
>     sys.exit(main())
>              ~~~~^^
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\cynthion\commands\cli.py", line 95, in main
>     args.func(device, args)
>     ~~~~~~~~~^^^^^^^^^^^^^^
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\cynthion\commands\cynthion_update.py", line 20, in cynthion_update
>     flash_bitstream(device, find_cynthion_bitstream(device, "analyzer.bit"))
>     ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\cynthion\commands\util.py", line 132, in flash_bitstream
>     programmer.flash(bitstream, offset=0)
>     ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\apollo_fpga\ecp5.py", line 735, in flash
>     self._flash_write_page(address, chunk)
>     ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\apollo_fpga\ecp5.py", line 683, in _flash_write_page
>     self._flash_wait_for_completion()
>     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\apollo_fpga\ecp5.py", line 603, in _flash_wait_for_completion
>     flash_status = self._background_spi_transfer([self.FlashOpcode.READ_STATUS1, 0])
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\apollo_fpga\ecp5.py", line 1089, in _background_spi_transfer
>     response = self.chain.shift_data(tdi=jtag_ready_data, length=bits_to_send, ignore_response=ignore_response)
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\apollo_fpga\jtag.py", line 479, in shift_data
>     response = self._shift_while_in_state('DRSHIFT', tdi=tdi, length=length, ignore_response=ignore_response,
>             byteorder=byteorder)
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\apollo_fpga\jtag.py", line 429, in _shift_while_in_state
>     result = self._scan_data(bit_length, data_bytes)
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\apollo_fpga\jtag.py", line 321, in _scan_data
>     result = self._scan_data_chunk(bits_in_chunk, chunk, ignore_response, advance_state)
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\apollo_fpga\jtag.py", line 367, in _scan_data_chunk
>     self.debugger.out_request(REQUEST_JTAG_SCAN, value=bits_to_scan, index=flags)
>     ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\apollo_fpga\__init__.py", line 369, in out_request
>     return self.device.ctrl_transfer(request_type, number, value, index, data, timeout=timeout)
>            ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\usb\core.py", line 1093, in ctrl_transfer
>     ret = self._ctx.backend.ctrl_transfer(
>                                 self._ctx.handle,
>     ...<4 lines>...
>                                 buff,
>                                 self.__get_timeout(timeout))
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\usb\backend\libusb1.py", line 897, in ctrl_transfer
>     ret = _check(self.lib.libusb_control_transfer(
>                                     dev_handle.handle,
>     ...<5 lines>...
>                                     length,
>                                     timeout))
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\usb\backend\libusb1.py", line 604, in _check
>     raise USBError(_strerror(ret), ret, _libusb_errno[ret])
> usb.core.USBError: [Errno 2] Entity not found

How do I fix it?
By the way, this is the information I got by " cynthion.exe info --force-offline"

> PS C:\Windows\System32> cynthion.exe info --force-offline
> Cynthion version: 0.2.2
> Apollo version: 1.1.1
> Python version: 3.13.5 (tags/v3.13.5:6cb20a2, Jun 11 2025, 16:15:46) [MSC v.1943 64 bit (AMD64)]
> 
> Forcing offline.
> 
> Found Cynthion device!
>         Hardware: Cynthion r1.4
>         Manufacturer: Great Scott Gadgets
>         Product: Cynthion Apollo Debugger
>         Serial number: FNE5CP2MGJJVCIBAEA3DQ.....
>         Vendor ID: 1d50
>         Product ID: 615c
>         bcdDevice: 0104
>         ADC reading: 3235
>         Firmware version: v1.1.1
>         USB API version: 1.2
>         Flash UID: 33612acba3a.....

Thanks.

tzuyin, 2025-08-04:
Hi @gniezen ,

Thanks for your suggestion. I extended the timeout duration, but it did not resolve the issue with my device. Below is the output I received when running the command to update the microcontroller firmware.
It appears to be a strange issue: **> OSError: exception: access violation writing 0x0000000000000000**

> PS C:\Windows\System32> cynthion.exe update --mcu
> Updating device firmware with 13888 bytes...
> Operation complete!
> Traceback (most recent call last):
>   File "C:\Program Files\Python313\Lib\weakref.py", line 666, in _exitfunc
>     f()
>     ~^^
>   File "C:\Program Files\Python313\Lib\weakref.py", line 590, in __call__
>     return info.func(*info.args, **(info.kwargs or {}))
>            ~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\usb\_objfinalizer.py", line 102, in _do_finalize_object_ref
>     obj._do_finalize_object()
>     ~~~~~~~~~~~~~~~~~~~~~~~^^
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
>     self._finalize_object()
>     ~~~~~~~~~~~~~~~~~~~~~^^
>   File "C:\Users\tchen15\AppData\Roaming\Python\Python313\site-packages\usb\backend\libusb1.py", line 616, in _finalize_object
>     _lib.libusb_unref_device(self.devid)
>     ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
**> OSError: exception: access violation writing 0x0000000000000000**

martinling, 2025-08-04:
@tzuyin Your latest attempt was actually successful, from these lines:

> PS C:\Windows\System32> cynthion.exe update --mcu
> Updating device firmware with 13888 bytes...
> Operation complete!

The subsequent crash occurs while the update program is exiting, having already done its job.

It's a bug when deconstructing some of the libusb objects used by pyusb - it ends up trying to free the same device object twice, because it's gotten mixed up between two very similar-looking appearances of the same Cynthion on the USB bus while the program was running.

This issue is being tracked at https://github.com/greatscottgadgets/pyfwup/issues/14