greatscottgadgets/cynthion #247: Access user_pmod A from gateware

lab313ru, 2025-08-25:
Hello,

I'm trying to access a `user_pmod A` from a gateware as it described in https://cynthion.readthedocs.io/en/latest/getting_started_facedancer_user_io.html but I cannot instantiate `Cynthion()`, but it says:
```python=
# cynthion/cynthion/python/src/cynthion.py", line 50, in Cynthion
board = CynthionBoard.autodetect(**board_identifiers)
# pygreat/board.py", line 81, in autodetect
raise DeviceNotFoundError()
```

How can I access individual user_pmod_0 pins from the gateware?

lab313ru, 2025-08-25:
That is what I was looking for, thanks! I'm just trying to add a required logic to that issue: https://github.com/greatscottgadgets/cynthion/issues/246

antoinevg, 2025-08-26:
`oe` is global for the entire connector.

lab313ru, 2025-08-26:
So I cannot specify which pins are inputs, for example? Or do you mean oe.eq(2) will disable the first pin and enable the second one?

antoinevg, 2025-09-18:
I mean that you can set the entire connector to output or input but not single pins. The underlying issue is the way we've declared these in `platform.py`.

If you want individual control over directionality take a look at how I'm accessing the PMOD connectors in the Facedancer gateware to declare custom resources:

https://github.com/greatscottgadgets/cynthion/blob/68e58e73bd37150a1729b638599f8ffde28bc98e/cynthion/python/src/gateware/facedancer/top.py#L351-L375

Though this is getting quite deep into the finer points of how Amaranth works with resources so you might have a bit more luck asking the question upstream!

closed 2025-09-18 by antoinevg